ros-navigation/navigation2 #3342: SMAC lattice planner does not work properly for differential/rectangular robot.

Yousseflah, 2022-12-28:
## Bug report

**Required Info:**

- Operating System: **Ubuntu 22.04**
  - 
- ROS2 Version: **Humble**
  - 
- Version or commit hash: **1.1.3 (from binaries)**
  - 
- DDS implementation:**rmw_cyclonedds**
  - 

#### Steps to reproduce issue:
**_I am not sure if this is a bug or not !_**
Send goal to the robot with opposite orientation while allowing the reverse option
I am using RPP as controller with SMAC lattice planner, here is my params for the planner:
```
planner_server:
  ros__parameters:
    expected_planner_frequency: 20.0
    use_sim_time: False
    planner_plugins: ["GridBased"]
    GridBased:
      plugin: "nav2_smac_planner/SmacPlannerLattice"
      allow_unknown: true  
      tolerance: 0.05
      max_iterations: 1000000         
      max_on_approach_iterations: 1000  
      max_planning_time: 5.0          
      analytic_expansion_ratio: 3.5 
      analytic_expansion_max_length: 4.0 
      reverse_penalty: 1.0 
      change_penalty: 0.05 
      non_straight_penalty: 1.05
      cost_penalty: 2.0
      rotation_penalty: 1.0               
      retrospective_penalty: 0.015
      lattice_filepath: "path_to_file/output.json" 
      lookup_table_size: 20.0  
      cache_obstacle_heuristic: true
      allow_reverse_expansion: true
      smooth_path: False
```
and for the primitive config:

```
{
    "motion_model": "diff",
    "turning_radius": 1.0,
    "grid_resolution": 0.05,
    "stopping_threshold": 5,
    "num_of_headings": 16
}
```

#### Expected behavior
Robot to navigate in reverse mode, reach goal and rotate to the requested orientation.

#### Actual behavior
Robot is parking as a vehicle when it approaches the goal.
/home/proiox/Pictures/Screenshots/Screenshot from 2022-12-21 21-28-32.png

#### Additional information
Not sure if this is a bug or not, but it seems that the generated path is for ackermann even if `"motion_model": "diff"` in the config.

SteveMacenski, 2022-12-28:
It does not look like your pictures were properly attached to visualize

Yousseflah, 2022-12-28:
thanks, updated.

SteveMacenski, 2023-01-02:
This is proper behavior. Its a kinematically feasible planner so its going to generate feasible paths. If you generate the lattice file using the Diff mode to allow for in-place rotations, it may do so, but you may want to increase or decrease the inplace rotation weight to penalize that more or less. I have it set by default to rotate only when absolutely required, rather than opportunistically to prevent rotations mid-path execution and only doing so at the start/end of the path when in confined spaces where there are no movements that would be able to obtain that orientation.

This case has plenty of space to get to the proper orientation without rotating in place, so it doesn't. You can adjust this by reducing the penalty on in place rotations, but it is still not guaranteed. The analytic expansions may be taking over here to generate the path due to trivial visibility - which knows nothing about in place rotation options.

closed 2023-01-03 by SteveMacenski