ros-navigation/navigation2 #3346: Navigation repeatedly fails for the first few goals when using SLAM

VishalDhayalan, 2023-01-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
- Version or commit hash:
  - 1.1.3-1
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
Without changing anything from the defaults, run: ```ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False slam:=True autostart:=True```.
This launches all the nodes, gazebo and RViz perfectly. RViz shows an initial patch of map built from the scans got from the robot being in its initial pose as shown in the screenshot below (as expected).

[image]

Now (using the Nav2 Goal button) send goals to the robot. 

#### Expected behaviour
The robot plans a path to that goal and follows it, driving the robot to the goal location while growing the map via SLAM.

#### Actual behaviour
The robot simply spins in place, moves back a short distance (officially known as 'back up') and repeats this randomly, taking a good 10-15 secs to show a path to the goal (if at all) and then struggling to even start to follow the path. This continues to happen for the first few goals without fail almost **every time** I launch, until a certain point (anywhere from the 2nd to **5th** goal) at which it actually starts moving significantly following the planned path and the first map update is seen (where the map grows).

After this point, there are never any problems following successive reasonable goals, even many challenging ones.
I have provided a short video below also demonstrating this visually. Luckily in this run the navigation only failed for the first goal and almost failed for the second...

https://user-images.githubusercontent.com/27903325/210475664-ddf868d8-a452-4f92-97e2-992bda0b3b2f.mp4

I've also noticed the control server say "Invalid path, Path is empty." when the robot stops following the path (even tho a full path to the goal is shown). This even happens in the video above for the second goal, as shown in the terminal at 1:01.

[image]

Is it possible to make it follow the initial goals (from the very first goal) more reliably?
I don't see this behaviour mentioned anywhere in the docs or on online videos demonstrating the same example.

SteveMacenski, 2023-01-04:
Provide your parameters and logs when this occurs

VishalDhayalan, 2023-01-04:
The parameter file is the default `nav2_params.yaml` that `tb3_simulation_launch.py` uses, so I've just attached it below as a txt file. Please note that as mentioned above, I am calling the launch file with `headless:=False slam:=True autostart:=True` arguments.
 
[nav2_params.txt](https://github.com/ros-planning/navigation2/files/10344927/nav2_params.txt)

The log file of a run very similar to the one in the video is below. The first goal failed as usual and the second one almost failed again. `Invalid path, Path is empty.` occurs twice. These goals are very simple in terms of navigation and should basically be straight obstacle-free paths, but for some reason, it really struggles to form a path and navigate using just what it can see upon launch... I think only after it has moved enough (via seemingly arbitrary recoveries like backup and spins) for the map to update a bit by chance, does navigation start working flawlessly.

```
[INFO] [launch]: All log files can be found below /home/u2002367/.ros/log/2023-01-04-14-03-51-110529-elk1-12417
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [12430]
[INFO] [gzclient-2]: process started with pid [12432]
[INFO] [spawn_entity.py-3]: process started with pid [12434]
[INFO] [robot_state_publisher-4]: process started with pid [12436]
[INFO] [rviz2-5]: process started with pid [12438]
[INFO] [component_container_isolated-6]: process started with pid [12441]
[INFO] [map_saver_server-7]: process started with pid [12462]
[INFO] [lifecycle_manager-8]: process started with pid [12464]
[INFO] [sync_slam_toolbox_node-9]: process started with pid [12476]
[robot_state_publisher-4] Link base_link had 7 children
[robot_state_publisher-4] Link camera_link had 2 children
[robot_state_publisher-4] Link camera_depth_frame had 1 children
[robot_state_publisher-4] Link camera_depth_optical_frame had 0 children
[robot_state_publisher-4] Link camera_rgb_frame had 1 children
[robot_state_publisher-4] Link camera_rgb_optical_frame had 0 children
[robot_state_publisher-4] Link caster_back_left_link had 0 children
[robot_state_publisher-4] Link caster_back_right_link had 0 children
[robot_state_publisher-4] Link imu_link had 0 children
[robot_state_publisher-4] Link base_scan had 0 children
[robot_state_publisher-4] Link wheel_left_link had 0 children
[robot_state_publisher-4] Link wheel_right_link had 0 children
[robot_state_publisher-4] [INFO] [1672841032.278271410] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-4] [INFO] [1672841032.278325511] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1672841032.278334147] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-4] [INFO] [1672841032.278339672] [robot_state_publisher]: got segment camera_depth_frame
[robot_state_publisher-4] [INFO] [1672841032.278344917] [robot_state_publisher]: got segment camera_depth_optical_frame
[robot_state_publisher-4] [INFO] [1672841032.278350289] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-4] [INFO] [1672841032.278355385] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher-4] [INFO] [1672841032.278360399] [robot_state_publisher]: got segment camera_rgb_optical_frame
[robot_state_publisher-4] [INFO] [1672841032.278365545] [robot_state_publisher]: got segment caster_back_left_link
[robot_state_publisher-4] [INFO] [1672841032.278370576] [robot_state_publisher]: got segment caster_back_right_link
[robot_state_publisher-4] [INFO] [1672841032.278375735] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-4] [INFO] [1672841032.278380878] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-4] [INFO] [1672841032.278385927] [robot_state_publisher]: got segment wheel_right_link
[map_saver_server-7] [INFO] [1672841032.306464911] [map_saver]: 
[map_saver_server-7] 	map_saver lifecycle node launched. 
[map_saver_server-7] 	Waiting on external lifecycle transitions to activate
[map_saver_server-7] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_saver_server-7] [INFO] [1672841032.306539925] [map_saver]: Creating
[lifecycle_manager-8] [INFO] [1672841032.308509813] [lifecycle_manager_slam]: Creating
[lifecycle_manager-8] [INFO] [1672841032.310058280] [lifecycle_manager_slam]: [34m[1mCreating and initializing lifecycle service clients[0m[0m
[lifecycle_manager-8] [INFO] [1672841032.310650445] [lifecycle_manager_slam]: [34m[1mStarting managed nodes bringup...[0m[0m
[lifecycle_manager-8] [INFO] [1672841032.310682249] [lifecycle_manager_slam]: [34m[1mConfiguring map_saver[0m[0m
[sync_slam_toolbox_node-9] [INFO] [1672841032.322162308] [slam_toolbox]: Node using stack size 40000000
[sync_slam_toolbox_node-9] [INFO] [1672841032.337709620] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-9] [INFO] [1672841032.338198603] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[map_saver_server-7] [INFO] [1672841032.339721648] [map_saver]: Configuring
[lifecycle_manager-8] [INFO] [1672841032.341413375] [lifecycle_manager_slam]: [34m[1mActivating map_saver[0m[0m
[map_saver_server-7] [INFO] [1672841032.341862409] [map_saver]: Activating
[map_saver_server-7] [INFO] [1672841032.341888961] [map_saver]: Creating bond (map_saver) to lifecycle manager.
[... cut ...]
[ERROR] [component_container_isolated-6]: process[component_container_isolated-6] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [component_container_isolated-6]: sending signal 'SIGKILL' to process[component_container_isolated-6]
[ERROR] [component_container_isolated-6]: process has died [pid 12441, exit code -9, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_isolated --ros-args --log-level info --ros-args -r __node:=nav2_container --params-file /tmp/tmpibfqukva --params-file /tmp/launch_params_dsbrtrv3 -r /tf:=tf -r /tf_static:=tf_static'].
```

SteveMacenski, 2023-01-04:
Huh, this is odd. I can see what you see. At first I thought it was due to some setting I accidentally flipped about tracking / handling unknown space (since there is unknown space you're trying to navigate to if the SLAM session hasn't viewed it yet) but instead I found something more interesting 

[image]

There's a block in the map right below the robot that's being picked up by SLAM. So that seems like actually a legit failure. Neat, haven't seen this bug before.

Looking at this situation, its odd because I don't see any laser scan points that end up there. But what is interesting is the fact that I definitely see laser scan points hitting the center pillar in the 9 who's line traced from the laser scanner origin align with that block of cost under the robot. 

Echoing the laser scan message I see a minimum value of 0.12 and max of 20m (slam toolbox has default of 20 too). 

If I change the turtlebot3's minimum laser scan range from 0.12 to 0.0 (which I changed maybe a few months ago due to self-occlusions of the model) things work again. Suspiciously, the blocks that exist seem to be about 0.1-0.15 from the robot's center (e.g. 2-3 grid cells) corresponding to that distance, so it seems that those values are bring thresholded to the minimum values. 

I believe the self occlusion issue has been resolved in the URDF, so I can revert the 0.12 to 0.0 to resolve this ticket w.r.t. Nav2, but for SLAM Toolbox, I'm not entirely sure from my ~45 minutes of debugging why it is doing that. 

https://github.com/ros-planning/navigation2/pull/3348 resolves. Thanks for the note!

closed 2023-01-04 by SteveMacenski

VishalDhayalan, 2023-01-04:
@SteveMacenski Thank you for investigating this! I had noticed that block but being completely new to ROS and everything here, I wasn't sure what to make of it.

So just to make sure I understand, since the laser had a minimum range set to 0.12, cells between 0.0 and 0.12 were treated as occupied even though the laser points originate from there and detect obstacles much further? It seems physically wrong to assume that in computation... Is that something SLAM Toolbox is doing?

Furthermore, even if the local map was perceived to have this block, why doesn't Nav2 follow the path it has already found around this ghost block?

Please help me understand the observed behaviour a bit better. Thank you.

VishalDhayalan, 2023-01-05:
Interestingly, the issue doesn't happen at all if we use the default turtlebot3 burger model instead of the waffle by running `ros2 launch nav2_bringup tb3_simulation_launch.py robot_sdf:=/opt/ros/humble/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf autostart:=True headless:=False slam:=True` as shown in the screenshot:

[image]

So I think it is something to do with the turtlebot3_waffle model... especially since the turtlebot3 burger still uses the minimum range of 0.12 for the lidar.

@SteveMacenski Do you think this might warrant further investigation?