ros-navigation/navigation2 #3347: logging and displaying time taken for completing the waypoint following

padhupradheep, 2023-01-04:
## Feature request

Follow up from the discussion in https://github.com/ros-planning/navigation2/pull/3165#discussion_r1060055035

#### Feature description

- Currently, we do not show how long did the robot took to complete the entire waypoint following. We just show the time taken for a specific waypoint and reset the time and move to the next. Maybe we need to have "total time taken" for showing how long the robot took to complete. 
- Additionally have an info stating about the "time taken for the last waypoint".
- Also I think it would be a nice idea for logging the time taken for each waypoint, which might be used for some sort of analysis later.

SteveMacenski, 2023-01-04:
Yes, so the waypoint follower result should store time to completion and return that on the final waypoint following result. 

The waypoint follower feedback can also contain the current total execution time alongside the current waypoint

Should be simple enough :smile:

SteveMacenski, 2023-01-12:
@padhupradheep was this on your radar to do?

padhupradheep, 2023-01-12:
Yes, it is! I need some time to get in here.. 

But if someone is already interested in getting their hands dirty, I'll let them take it.

Ryanf55, 2023-01-19:
Hey there, I'd like to learn a bit about waypoints in NAV2. Since this seems like the most reasonably scoped issue related to waypoints, mind assigning it to me?

SteveMacenski, 2023-01-19:
Sure! We'd love the help here and I think its a great well-defined scope project!

Ryanf55, 2023-01-20:
> Sure! We'd love the help here and I think its a great well-defined scope project!

Awesome I appreciate it. I'll need a bit of time to get up to speed. First off, it appears there to be support for pausing the navigation, or [waiting](https://github.com/ros-planning/navigation2/blob/main/nav2_msgs/action/Wait.action). 

When computing the total time to complete a waypoint set, should the time the robot is "waiting" or "paused" be excluded from the elapsed time? 

IE, if robot pauses, elapsed timer temporarily is paused.

SteveMacenski, 2023-01-20:
No, I think the full time it actually takes, even when completing some task or recovery behavior operations should count

Ryanf55, 2023-01-20:
It took me a bit of time to figure out how to use the waypoint follower and lots of squinting at the discourse Gif's, but I figured out how to use it. 

On my branch, it's printing the time to completion successfully. 
`[component_container_isolated-6] [INFO] [1674254602.141816559] [waypoint_follower]: Completed all 5 waypoints requested in a time of 49.68 seconds.`

Now, for the UI.. Here's a screenshot of it currently while in WP mode:
[image]

First off, can I clear up requirements? Please approve:
1) Change name of "Time Taken" to "Time Taken for Current WP"
2) Add another text box below Time Taken for "Total Time Taken"

Right now, I can get box 2 to update when the waypoints are complete/aborted/started easily, but did you also want it to live-update as the time increases?

SteveMacenski, 2023-01-20:
> Completed all 5 waypoints requested in a time of 49.68 seconds.

Awesome! I think it makes sense to have the feedback message of the waypoint follower have the current task time reported through it so the panel can have that information. I think @padhupradheep also mentioned that may be the Action result can have a `duration []` of times for each waypoint for later analysis (not used in the panel, but if you're making time tracking tools, fits right in). 

I think there are examples in the rviz plugin already of a particular task adding in or replacing fields in the feedback part of the panel? If so, I think you can leave the default view the same, but for WPF, you can change that (1) you mention and add (2).

Resources
- https://github.com/ros-planning/navigation2/blob/main/nav2_rviz_plugins/src/nav2_panel.cpp#L798-L807
- https://github.com/ros-planning/navigation2/blob/main/nav2_rviz_plugins/src/nav2_panel.cpp#L1479-L1507

Ryanf55, 2023-01-21:
You got it.

Can we define behavior for the result `duration [] total_elapsed_time` and how it's built, particularly for preempts? Here's my proposal with an example where the user calls follow waypoint action with 3 waypoints with 1 loop. Would this align with user expectations?

**Happy Path Example:** 

Robot starts at t=A, then at the instant it completes the first waypoint 0 at t=B:
`result.total_elapsed_time = [B-A]`
Then, once it hits waypoint 1 at time C:
`result.total_elapsed_time = [B-A, C-A]`
And, so on until
`result.total_elapsed_time = [B-A, C-A, D-A, E-A, F-A, G-A] `
Where `len(total_elapsed_time) = len(goal.poses) * goal.number_of_loops`, so 6 in this case.

Continuously, feedback.current_elapsed_time is T-A, where T is the current time. 

**Preempt Example:**
Assuming the above is true, this is what I propose in the case the follow waypoints action is preempted. 

Robot starts at t=A, then at the instant it completes the first waypoint 0 at t=B:
`total_elapsed_time = [B-A]`
Then, halfway to waypoint 1, it gets preempted somehow at time X. 
`total_elapsed_time = [B-A, X-C]`
To know waypoint 1 was never reached in the action client, you can query it in 
`result.missed_waypoints = [{ .index=1}] `

The key thing is, just like `missed_waypoints`, the size of total_elapsed_time depends on if any waypoints were missed.

SteveMacenski, 2023-01-21:
total time elapsed is not a vector, its just the time elapsed. The result message with time for each waypoint would just be the time component of that particular waypoint sub-task `N - N-1`. But yes, with loops, then the size would be `N * M` for N waypoints over M loops.

Preempting is a new request. All gets reset.

Ryanf55, 2023-01-21:
Sorry, I'm not following. Can you clarify your proposed `FollowWaypoints.action` interface? 

Right now my branch is: 
```
# Error codes
# Note: The expected priority order of the errors should match the message order
uint16 NONE=0
uint16 UNKNOWN=600
uint16 TASK_EXECUTOR_FAILED=601

#goal definition
uint32 number_of_loops
uint32 goal_index 0
geometry_msgs/PoseStamped[] poses
---
#result definition
MissedWaypoint[] missed_waypoints
builtin_interfaces/Duration total_elapsed_time
---
#feedback definition
uint32 current_waypoint
```

SteveMacenski, 2023-01-21:
```

#goal definition
uint32 number_of_loops
uint32 goal_index 0
geometry_msgs/PoseStamped[] poses
---
#result definition
MissedWaypoint[] missed_waypoints
builtin_interfaces/Duration[] wp_elapsed_time
---
#feedback definition
uint32 current_waypoint
builtin_interfaces/Duration elapsed_time
```

padhupradheep, 2023-01-23:
> think @padhupradheep also mentioned that may be the Action result can have a duration [] of times for each waypoint for later analysis

Yes, we could also make logging as optional (a check box in RViz - disabled by default), so that for the users who are just up for some quick prototyping and are not interested in the timing. If @SteveMacenski  is open for adding another field to the `FollowWaypoints` action, I would propose adding something like `log_wp_elapsed_time` (of course boolean). By this way, if an user is using the waypoint following from his own application by calling the waypoint server, he gets the option if he want to log the waypoints or not. 

Also while logging, I would also find a way to incorporate the missed waypoints. So that it would be easy for the users to check and process the data accordingly. I'm just imagining a text file something like:

```
waypoint  time (s) status (1 - reached, 0 - missed )
0               2            1
1               4            1
2               3            0
0               2            1
1               4.12         0
2               2            1
```

Ryanf55, 2023-01-23:
I'm actually confused why there needs to be server-side logging. It seems like the client already have access to the time information. 
If that's true, why does it need to be part of the action interface and increase complexity of the message with some fields that are only sometimes used. If the time the action took was important, wouldn't the client need to implement it for all the actions it is calling, instead of just this one action? Unless the plan is to add timing information to ALL the actions.

SteveMacenski, 2023-01-23:
> log_wp_elapsed_time

Why? I don't see why we wouldn't do so. Its just 1 more field in the rviz panel? Half of the info in the feedback panel probably isn't being looked at, but its still provided :laughing: I try to not have unnecessary configurations / parameters where its not necessary. It adds complication (I'm hoping to have about 2 weeks this year to be able to audit all of Nav2 and throw out as many as possible).

Why would the missed waypoints and the time for each waypoint to complete be related data to structure together?

> If the time the action took was important, wouldn't the client need to implement it for all the actions it is calling, instead of just this one action?

Many do have it already. The controller server reports time elapsed. The planner server reports time to complete planning.

padhupradheep, 2023-01-23:
I was thinking in a way such that, what if there are 60 waypoints that are looped 60 times. Randomly the 7th waypoint fails 7 times, each failure was some unexpected issue such as humans or forklifts or any other components blocking the robot goal, which blocked the robot multiple times on a specific waypoint before which it failed and considered to be missed. I thought user might have this additional data for the 7th waypoint where he see’s that it failed frequently that too at different time durations. Maybe this could give the user additional input on this problematic waypoint and maybe he can do some sort of analysis: maybe a physical one or if he know that it’s already a crowded region then he could shift the waypoint to somewhere else. 
 
 Also the reason for adding a switch for the logs was to do some sort of memory management (I know it’s not going to take a lot of space but still😅), so that we don’t have to create a text file for each run. 
 
 I feel, I might be bit overthinking (just with the motive to make Nav2 better). I think for now, we could just stick with the initial idea 😄

SteveMacenski, 2023-01-23:
> so that we don’t have to create a text file for each run.

... why would we be making text files?

padhupradheep, 2023-01-23:
Don’t worry, t’s just a continuation of the idea what I had built in my mind.

Ryanf55, 2023-01-23:
Can I limit the scope to original intention, and save text file saving to another issue?

SteveMacenski, 2023-01-23:
That's not part of this ticket - just the provision of this information from the waypoint follower and changing the rviz plugin to use the feedback total time / waypoint times.

Ryanf55, 2023-01-25:
While I'm working on the UI, should I add a display for the current waypoint and/or loop number? Right now there it's not clear how complete the task is. 

[image]

In Ardupilot Mission Planner for example, on the right bottom it shows a display of distance to next waypoint and the current waypoint number. 
[image]
See `13m/2`. For a long running mission, it is nice to show that.

SteveMacenski, 2023-01-25:
That is already happening if you look at the panel today! The Nav to Pose interface provides an ETA, distance to goal. 

> current waypoint and/or loop number

I generally would prefer to review 2 smaller PRs over 1 larger PRs when there can be a clear separation. So I think that's a great idea and would love to see that represented, but I think that's for the next PR :smile:

Ryanf55, 2023-01-25:
Sounds good, but it doesn't show current waypoint number; I'll create a follow up issue for that. 

Things are going well otherwise, aside from realizing there was nothing subscribed to the feedback, and that all the waypoint live data was coming from a different client and is stored in a different section of the code than the waypoint callbacks. Since the message content is the similar, it's a bit unclear as an application developer which one to use. I might need to put this up for draft for some input if I can't figure it out.

SteveMacenski, 2023-01-25:
The current waypoint number is in the current feedback message of the waypoint follower (the same feedback you're modifying to get the full navigation time of the WPF task) 

OK! The rviz2 plugin admittedly is a little bit of a bastardized use of actions in some regards. Using the private topics of the action server internal isn't really a best practice, but the rviz panel requires a little but of creative accounting. I usually wouldn't accept that in live-running code, but the rviz panel is mostly for testing and demos, not deployed applications, so I give it a slide under the spirit of legacy code before my time (that keeps growing...)

Ryanf55, 2023-01-25:
[image]
Cool, got it working, PR up in a bit. Before merge, I'd really like to add a few tests to the waypoint follower code.

padhupradheep, 2023-01-25:
Awesome, really cool.. A working GIF of the same would be nice. 

Just a small suggestion, You don't have to go 6 decimal places for the time. I'd probably round it up milliseconds.

> like to add a few tests to the waypoint follower code.

I know someone who'll be very happy to have the tests and see in the increase of the code coverage.. :100:

Ryanf55, 2023-01-25:
> Awesome, really cool.. A working GIF of the same would be nice.
> 
> Just a small suggestion, You don't have to go 6 decimal places for the time. I'd probably round it up to maybe milliseconds.
> 
> > like to add a few tests to the waypoint follower code.
> 
> I know someone who'll be very happy to have the tests and see in the increase of the code coverage.. 100

Done https://github.com/ros-planning/navigation2/pull/3384/commits/91509c3ab8044703fc5279861f3a29f9d92c1024

https://user-images.githubusercontent.com/25047695/214632010-7af6809e-3c94-40f9-bee8-bb81846c73dd.mp4

Ryanf55, 2023-01-25:
[image]
The gif converter messed up the color a bit, but you should get the point.

SteveMacenski, 2024-06-11:
Closing as `wontfix`, this doesn't seem to be of much interest and the technical challenges are not worth the superficial payoff. If someone would like to implement / needs it, I'd welcome a functional PR to review and merge, but this can be safely closed.

closed 2024-06-11 by SteveMacenski