ros-navigation/navigation2 #3355: The value of server_timeout_ of on_new_goal_received() in bt_action_node.hpp is too short.

tosa-no-onchan, 2023-01-10:
## Bug report

**Required Info:**

- Operating System: 
  - Ubuntu 20.04
- ROS2 Version:
  - foxy and source build from github.
- Env:
  - Jetson nano 2G (Jet Pack 4.X and ubuntu 20.04) and rtabmap_ros

#### Steps to reproduce issue

```
on SBC(Jetson nano 2G), run rtabmap_ros with Single USB Stereo Camara(stereo_sync) 、GPS、robot_localization、 and navigation2.

stereo_image_proc.launch.py
rtabmap_ros/stereo_sync
rtabmap_ros/point_cloud_xyz
robot_localization/navsat_transform_node
robot_localization/ekf_node
rtabmap_ros/rtabmap
too heavy load.
```

#### Expected behavior
"Navigation failed" not appear

#### Actual behavior
[bt_navigator-4] [INFO] [1673251680.513935747] [bt_navigator]: Begin navigating from current location to (1.39, -0.08)
[bt_navigator-4] [ERROR] [1673251680.548460478] [BehaviorTreeEngine]: Behavior tree threw exception: send_goal failed. Exiting with failure.
[bt_navigator-4] [ERROR] [1673251680.649125020] [bt_navigator]: Navigation failed
[bt_navigator-4] [WARN] [1673251680.649162833] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.

----
## Feature request
The value of server_timeout_ of on_new_goal_received() in bt_action_node.hpp is too short.
I would like to increase a value of  server_timeout_ on the follwoing code,
bt_action_node.hpp
lines 233
    if (rclcpp::spin_until_future_complete(node_, future_goal_handle, server_timeout_) !=
      rclcpp::FutureReturnCode::SUCCESS)

I had tested the follwoing code,
    //if (rclcpp::spin_until_future_complete(node_, future_goal_handle, server_timeout_) !=
    //  rclcpp::FutureReturnCode::SUCCESS)
    // test by nishi
    if (rclcpp::spin_until_future_complete(node_, future_goal_handle, server_timeout_*2) !=
      rclcpp::FutureReturnCode::SUCCESS)
    {
      throw std::runtime_error("send_goal failed");
    }

It everything goes OK.

How do I set the value of  "server_timeout" from parameter file.
lines 41
    // Get the required items from the blackboard
    server_timeout_ =
      config().blackboard->template get<std::chrono::milliseconds>("server_timeout");
    getInput<std::chrono::milliseconds>("server_timeout", server_timeout_);

SteveMacenski, 2023-01-10:
> too heavy load.

That seems like the root of the problem, no? What's your CPU / memory at when this happens? Should you not try instead to reduce the load by either using a lighter weight set of software or a more capable piece of hardware?

But yes, you can increase that timeout rather than modifying any code, so that should be easy. You just need to set the `server_timeout` port on the BT nodes where you have timing issues which use the BT Action Node (probably FollowPath and ComputePathToPose) in your BT XML.

closed 2023-01-10 by SteveMacenski

tosa-no-onchan, 2023-01-11:
Thank you.

I think, setting parameter is in here.

From nav2_params.yaml file definition, BT XML is navigate_w_replanning_and_recovery.xml and global plan part is ComputePathToPose.

bt_navigator:
  ros__parameters:
    default_bt_xml_filename: "navigate_w_replanning_and_recovery.xml"

https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_bt_navigator/behavior_trees/navigate_w_replanning_and_recovery.xml

&lt;RecoveryNode number_of_retries="1" name="ComputePathToPose"&gt;
&lt;!-- &lt;ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/&gt; --&gt;
&lt;ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased" server_timeout="20" /&gt;