ros-navigation/navigation2 #3361: On Abort controller server does not publish zero velocity

padhupradheep, 2023-01-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Rolling
- Version or commit hash:
  - recent
- DDS implementation:
  - Cyclone
#### Steps to reproduce issue

I was testing the MPPIC controller and for some reason at a certain point during the navigation the controller failed with the logs:

```
[controller_server-4] [ERROR] [1673626999.321900985] [controller_server]: Action server failed while executing action callback: "Resulting plan has 0 poses in it."
[controller_server-4] [WARN] [1673626999.321987554] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[bt_navigator-7] [ERROR] [1673626999.356187680] [bt_navigator]: Goal failed
[bt_navigator-7] [WARN] [1673626999.356250669] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle

```
but the issue was that, the robot didn't stop moving when it reached this state. It was continuing with the last velocity command. I had stop the process, due to safety reason. 

In general when the controller server aborts, it should `publishZeroVelocity` right?

#### Expected behavior
when the controller server aborts, it should `publishZeroVelocity`

#### Actual behavior
It is not publishing zero velocity during abort

SteveMacenski, 2023-01-13:
> Resulting plan has 0 poses in it

I see that this exception is thrown from within the controller plugins RPP/DWB/MPPI, not from the controller server itself. 

> throw std::runtime_error

Interstingly, MPPI was developed pre-contextual error codes from @jwallace42 so that's actually a really useful oversight recognition @padhupradheep that I need to update these exceptions to be semantically meaningful. 

> Action server failed while executing action callback

That's from the simple action server catching a generic `std::exception` if the callback does not catch its own exceptions.

---

So, my 3 minutes of code-doctoring says that the issue here is that the controller server doesn't have a generic `std::runtime_error` catch in the try/catch/catch section so the Simple Action Server is pulling the slack. As a result, `publishZeroVelocity` isn't called because the simple action server knows nothing about the server being used. 

Fix: Add `std::runtime_error` catch to controller server. Planner Server has the `std::exception` catch, but Smoother Server is also missing a base-class exception catch for those not using the new error codes (historical codebases, folks that threw custom exception). 

Pretty straight forward :-) 10 line change

padhupradheep, 2023-01-13:
> The MPPI stuff is not the cause of this issue

Of course not, I too suspected it somewhere in the controller server.. 

------------------------------------------------------
but I know I'm deviating from the issue, just wanted to ask when we get this error

> [controller_server-4] [ERROR] [1673626999.321900985] [controller_server]: Action server failed while executing action callback: "Resulting plan has 0 poses in it."

does this mean that, the optimizer wasn't able to find the solution ?

SteveMacenski, 2023-01-17:
https://github.com/ros-planning/navigation2/pull/3373 resolves

closed 2023-01-17 by SteveMacenski