ros-navigation/navigation2 #3363: Build warning: easy_install and setup.py install are deprecated

AndyZe, 2023-01-14:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22
- ROS2 Version:
  - Rolling, building from source
- Version or commit hash:
  - latest commit on main (c841417)
- DDS implementation:
  - Cyclone

When building from source, I get these warnings:

```
#0 46.17 --- stderr: nav2_simple_commander
#0 46.17 /usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
#0 46.17   warnings.warn(
#0 46.17 /usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
```

SteveMacenski, 2023-01-17:
Does that not happen for all python ROS 2 packages? The last my understanding was, this wasn't something we could migrate away from (yet?) out of our control. If that's changed, it would be good to know and happy to have fixed in Nav2!

AlexeyMerzlyakov, 2023-01-17:
I have the same issue on Jammy and ROS2 Rolling built from sources. No problems on Focal with the same ROS2 sources. 
There are also lots of same messages during ROS2 build process itself. Seems to have the same problems described in https://github.com/colcon/colcon-core/issues/454 and https://github.com/ament/ament_cmake/issues/382. Both tickets are open, as a workaround they are suggesting to downgrade the `setuptools` python package version (although did not checked yet locally w/a).

SteveMacenski, 2023-01-18:
Until there's a real ROS-2-wide solution, I'm going to close this since its not actionable.

closed 2023-01-18 by SteveMacenski