ros-navigation/navigation2 #3364: Build error: undefined reference to `rclcpp::PublisherBase::PublisherBase

AndyZe, 2023-01-14:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22
- ROS2 Version:
  - Rolling, building from source
- Version or commit hash:
  - latest commit on main (c841417)
- DDS implementation:
  - Cyclone

When building from source, I get this error:

```
#0 71.22 --- stderr: nav2_util
#0 71.22 /usr/bin/ld: /opt/ros/rolling/lib/libbondcpp.so: undefined reference to `rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_message_type_support_t const&, rcl_publisher_options_s const&, rclcpp::PublisherEventCallbacks const&, bool)'
#0 71.22 /usr/bin/ld: /opt/ros/rolling/lib/libbondcpp.so: undefined reference to `rclcpp::SubscriptionBase::SubscriptionBase(rclcpp::node_interfaces::NodeBaseInterface*, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_s const&, rclcpp::SubscriptionEventCallbacks const&, bool, bool)'
#0 71.22 /usr/bin/ld: /opt/ros/rolling/lib/libbondcpp.so: undefined reference to `rclcpp::Clock::now() const'
#0 71.22 collect2: error: ld returned 1 exit status
```

I think the docker commands I'm using to build are pretty straight forward:

```
# Install dependencies
# Note the ros-rolling-gazebo-ros* packages are needed here as they are 
# not captured by the package dependencies when running rosdep install.
RUN apt-get update && \
    apt-get install --no-install-recommends -y \
        bash-completion \
        openssh-client \
        python3-pip \
        ros-rolling-gazebo-ros*

# Set up CycloneDDS as ROS 2 middleware
RUN apt-get update && \
    apt-get install --no-install-recommends -y \
        ros-rolling-rmw-cyclonedds-cpp
ENV RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

# Create Underlay workspace
ARG UNDERLAY_WS=/opt/underlay_ws
ENV UNDERLAY_WS=${UNDERLAY_WS}
WORKDIR ${UNDERLAY_WS}/src
COPY ./tiago_simulation.repos ../
RUN vcs import < ../tiago_simulation.repos
RUN source /opt/ros/rolling/setup.bash && \
    rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
WORKDIR ${UNDERLAY_WS}
RUN source /opt/ros/rolling/setup.bash && \
    colcon build --symlink-install --packages-skip nav2_system_tests
```

AndyZe, 2023-01-14:
Link to the definition of PublisherBase:  https://github.com/ros2/rclcpp/blob/3db2ece14545ade3ab9cec0146be57c1a3ae83f8/rclcpp/include/rclcpp/publisher_base.hpp#L77

SteveMacenski, 2023-01-17:
> /opt/ros/rolling/lib/libbondcpp.so

This is not Nav2. Have you updated your binaries for Bond?

closed 2023-01-17 by SteveMacenski

sea-bass, 2023-01-22:
There may be something going on here... because I can reproduce this build error when building `nav2_util` from source on an image based off `ros:rolling`.

... but if I additionally bring in https://github.com/ros/bond_core/tree/ros2 as a source build (not from binary), things are fine.

Are you folks perhaps also building this package from source?

SteveMacenski, 2023-01-23:
Note that I did actually run into this the other day and it was caused by cached parts for re-builds in my workspace becoming invalid due to updates in binary packages. You have to clean your workspace and have a clean rebuild after pulling in new updates to Rolling binaries which change ABI because Rolling is not a stable distribution and ABI is subject to change.