ros-navigation/navigation2 #3367: Regulated Pure Pursuit Lookahead Point resets to start of the path instead of nearest point to robot if the path is larger than max of costmap dimensions

swarajpeppermint, 2023-01-17:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04
- ROS2 Version: Humble (Installation Method: Debian Packages)
- Version or commit hash: 1.1.3-1jammy.20221207.081347
- DDS implementation: CycloneDDS

#### Steps to reproduce issue
In a map, create a long winding path whose length will be greater than the max of the set costmap dimensions, and do not set the max_robot_pose_search_dist parameter for RPP. Create a condition such that RPP is cancelled at a distance on the path beyond the max of the set costmap dimensions (in our test case, we created a Behavior Tree which cancels RPP if there's an obstacle in front. When the obstacle is removed, it re-activates RPP giving it the original path)

#### Expected behavior
RPP should set the lookahead point to the point on path nearest to the robot before starting to give command velocities 

#### Actual behavior
If the robot's pose is at a distance on the path beyond the max of the costmap dimension (or distance set as the max_robot_pose_search_dist parameter), the lookahead point resets to the start of the path and RPP starts giving command velocities such that the robot starts trying to move to the start of the path.

[image]

#### Our Diagnosis
We diagnosed that this was due to the `closest_pose_upper_bound` variable in the `RegulatedPurePursuitController::transformGlobalPlan` method. If the _max_robot_pose_search_dist_ parameter is not specified, then it takes the max of the 2 costmap dimensions as the default parameter, and when the robot's distance travelled on the path exceeds that distance, it resets the lookahead point to the start.
Even if the _max_robot_pose_search_distance_ parameter is specified, if in any case the robot's distance travelled exceeds that distance, it will reset the lookahead point to the start of the path. 

#### Our Solution
We figured there are cases when the current code in RPP will be required. However, for cases where the robot will have to travel arbitrarily long distances, there needs to be a parameter to allow RPP to take the nearest lookahead point no matter where the robot is on the path.
For that, we created a _use_full_path_for_closest_pose_ parameter. If that parameter is enabled, the closest_pose_upper_bound is set as the global_plan.poses.end().

We wanted to put forth this solution to the Nav2 community to get some opinions whether this is a valid bug before raising a PR.
Please let me know if I've missed updating any required information in this issue.

SteveMacenski, 2023-01-17:
Is this because you're not replanning the path? 

>  If the max_robot_pose_search_dist parameter is not specified, then it takes the max of the 2 costmap dimensions as the default parameter

Yes, because typically the path is being replanned regularly so we only need to search within a restricted window size of the initial point to accommodate for distance traveled between planning iterations (and those that have valid cost values to check for -- actually existing on the local costmap). 

>  Even if the max_robot_pose_search_distance parameter is specified, if in any case the robot's distance travelled exceeds that distance, it will reset the lookahead point to the start of the path.

It is a parameter for a reason - if you don't set it properly for your situation, its obviously not going to work very well. It exists mostly for dealing with looping paths so that if you have a cross of the path over itself that you select the proper one instead of being basically a coin flip on really small amounts of floating point error. 

> there needs to be a parameter to allow RPP to take the nearest lookahead point no matter where the robot is on the path.

If you don't have that looping issue - you could set this to max numerical limits. I'm not sure this requires a new parameter, you can just set it to `1.79769e+308` (or something else, but that's just max value for doubles)

swarajpeppermint, 2023-01-17:
> Is this because you're not replanning the path?

Yep, we're not replanning the path.

> It exists mostly for dealing with looping paths so that if you have a cross of the path over itself that you select the proper one instead of being basically a coin flip on really small amounts of floating point error. 

Yep, this makes sense. We don't have the looping issue as of now, but hadn't thought of this. As an initial thought, maybe a possible approach to the looping issue could be keeping a buffer of the last few poses of the path on which the robot was present and using them as well in finding the nearest point. Here the assumption is that the robot can't just jump between paths and will maintain relative continuity of traversed points.

> you can just set max_robot_pose_search_distance to 1.79769e+308

Yeah that would work too. As a personal opinion, a defined parameter looked like a more readable approach. But I agree, a specific parameter for this might not be required.

SteveMacenski, 2023-01-18:
> As an initial thought, maybe a possible approach to the looping issue could be keeping a buffer of the last few poses of the path on which the robot was present and using them as well in finding the nearest point. 

That could work, though if you replan on any regularity, it would pretty much be `0` most of the time and you'd have to have a reset every time you replan to reset the buffer length to `0`. It would absolutely work, but not really be any simpler than what's possible today. But if you find that you have some looping problems + not-replanning, that sounds like a plausible solution that could be contributed!

> Yeah that would work too. As a personal opinion, a defined parameter looked like a more readable approach. But I agree, a specific parameter for this might not be required.

I don't disagree. We could also make some negative value `-1` be "disable" and have it set to `numeric_limits<double>::max()`, there's good precedence for that. That would be more natural looking than some arbitrarily high number.

swarajpeppermint, 2023-01-24:
Okay, seems like there's possibility for a small change here. I would like to contribute if possible. I haven't contributed to Nav2 before. What should be my next steps? Should I just go ahead and open a PR @SteveMacenski?
Sorry for the late reply. Was out of action last week.

SteveMacenski, 2023-01-24:
Open a PR would be the next step :-) I think setting it to `-1` (or a negative number) corresponding in the code to `numeric_limits<double>::max()` with a log about what's happening would be the most natural. Just make sure to also open a PR against our documentation to document that a negative value for this param will allow for the distance to search as infinite. Page: https://navigation.ros.org/configuration/packages/configuring-regulated-pp.html. Also, migration guide page entry about the change so that folks migrating to this distribution know about important features and changes.

SteveMacenski, 2023-02-22:
https://github.com/ros-planning/navigation2/pull/3424 resolves

closed 2023-02-22 by SteveMacenski