ros-navigation/navigation2 #3386: nav2_util issue build - main branch - humble distro

ionutnechita, 2023-01-27:
Ubuntu 22.04 - Humble distro ROS

#23 129.5 + colcon build
#23 130.2 Starting >>> nav2_common
#23 130.2 Starting >>> nav_2d_msgs
#23 130.3 Starting >>> nav2_gazebo_spawner
#23 130.5 /usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' instead
#23 130.5   warnings.warn(
#23 130.5 /usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
#23 130.5   warnings.warn(
#23 131.3 --- stderr: nav2_gazebo_spawner
#23 131.3 /usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' instead
#23 131.3   warnings.warn(
#23 131.3 /usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
#23 131.3   warnings.warn(
#23 131.3 ---
#23 131.3 Finished <<< nav2_gazebo_spawner [1.03s]
#23 131.6 Finished <<< nav2_common [1.33s]
#23 131.6 Starting >>> nav2_msgs
#23 131.6 Starting >>> nav2_voxel_grid
#23 138.6 Finished <<< nav_2d_msgs [8.32s]
#23 138.6 Starting >>> dwb_msgs
#23 146.5 Finished <<< nav2_voxel_grid [14.9s]
#23 151.8 Finished <<< dwb_msgs [13.2s]
#23 165.3 Finished <<< nav2_msgs [33.7s]
#23 165.3 Starting >>> nav2_util
#23 180.9 --- stderr: nav2_util
#23 180.9 In file included from /home/eiforamr/ros2_ws/src/navigation2/nav2_util/include/nav2_util/geometry_utils.hpp:24,
#23 180.9                  from /home/eiforamr/ros2_ws/src/navigation2/nav2_util/src/costmap.cpp:19:
#23 180.9 /opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: error: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Werror=cpp]
#23 180.9    35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
#23 180.9       |  ^~~~~~~
#23 180.9 In file included from /home/eiforamr/ros2_ws/src/navigation2/nav2_util/include/nav2_util/robot_utils.hpp:28,
#23 180.9                  from /home/eiforamr/ros2_ws/src/navigation2/nav2_util/src/robot_utils.cpp:20:
#23 180.9 /opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: error: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Werror=cpp]
#23 180.9    35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
#23 180.9       |  ^~~~~~~
#23 180.9 In file included from /opt/ros/humble/include/rclcpp/rclcpp/logging.hpp:24,
#23 180.9                  from /opt/ros/humble/include/rclcpp/rclcpp/client.hpp:40,
#23 180.9                  from /opt/ros/humble/include/rclcpp/rclcpp/callback_group.hpp:23,
#23 180.9                  from /opt/ros/humble/include/rclcpp/rclcpp/any_executable.hpp:20,
#23 180.9                  from /opt/ros/humble/include/rclcpp/rclcpp/memory_strategy.hpp:25,
#23 180.9                  from /opt/ros/humble/include/rclcpp/rclcpp/memory_strategies.hpp:18,
#23 180.9                  from /opt/ros/humble/include/rclcpp/rclcpp/executor_options.hpp:20,
#23 180.9                  from /opt/ros/humble/include/rclcpp/rclcpp/executor.hpp:37,
#23 180.9                  from /opt/ros/humble/include/rclcpp/rclcpp/executors/multi_threaded_executor.hpp:25,
#23 180.9                  from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:21,
#23 180.9                  from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
#23 180.9                  from /home/eiforamr/ros2_ws/src/navigation2/nav2_util/include/nav2_util/costmap.hpp:21,
#23 180.9                  from /home/eiforamr/ros2_ws/src/navigation2/nav2_util/src/costmap.cpp:17:
#23 180.9 /home/eiforamr/ros2_ws/src/navigation2/nav2_util/src/costmap.cpp: In constructor ‘nav2_util::Costmap::Costmap(rclcpp::Node*, bool, bool, int, int)’:
#23 180.9 /home/eiforamr/ros2_ws/src/navigation2/nav2_util/src/costmap.cpp:42:28: error: format ‘%f’ expects argument of type ‘double’, but argument 5 has type ‘int’ [-Werror=format=]
#23 180.9    42 |       node_->get_logger(), "Costmap: Lethal threshold set to %.2f, it should be within"
#23 180.9       |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
#23 180.9    43 |       " bounds 0-100. This could result in potential collisions!", lethal_threshold_);
#23 180.9       |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~~~~~~~
#23 180.9       |                                                                    |
#23 180.9       |                                                                    int
#23 180.9 /home/eiforamr/ros2_ws/src/navigation2/nav2_util/src/costmap.cpp:42:65: note: format string is defined here
#23 180.9    42 |       node_->get_logger(), "Costmap: Lethal threshold set to %.2f, it should be within"
#23 180.9       |                                                              ~~~^
#23 180.9       |                                                                 |
#23 180.9       |                                                                 double
#23 180.9       |                                                              %.2d
#23 180.9 cc1plus: all warnings being treated as errors
#23 180.9 gmake[2]: *** [src/CMakeFiles/nav2_util_core.dir/build.make:76: src/CMakeFiles/nav2_util_core.dir/costmap.cpp.o] Error 1
#23 180.9 gmake[2]: *** Waiting for unfinished jobs....
#23 180.9 cc1plus: all warnings being treated as errors
#23 180.9 gmake[2]: *** [src/CMakeFiles/nav2_util_core.dir/build.make:160: src/CMakeFiles/nav2_util_core.dir/robot_utils.cpp.o] Error 1
#23 180.9 gmake[1]: *** [CMakeFiles/Makefile2:212: src/CMakeFiles/nav2_util_core.dir/all] Error 2
#23 180.9 gmake: *** [Makefile:146: all] Error 2
#23 180.9 ---
#23 180.9 Failed   <<< nav2_util [15.7s, exited with code 2]
#23 181.0
#23 181.0 Summary: 6 packages finished [51.0s]
#23 181.0   1 package failed: nav2_util
#23 181.0   2 packages had stderr output: nav2_gazebo_spawner nav2_util
#23 181.0   24 packages not processed
------

SteveMacenski, 2023-01-27:
Use the humble branch for humble. Main is for Rolling

closed 2023-01-27 by SteveMacenski

ionutnechita, 2023-01-29:
Hi,

I used both branch and this issue is reproduced.