ros-navigation/navigation2 #3390: AMCL Couldn't determine robot's pose associated with laser scan

Jayden-F, 2023-02-01:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04
- ROS2 Version:
Galactic Binaries
- Version or commit hash:
1.0.12-1focal.20221208.082109

- DDS implementation:
Cyclone

#### Steps to reproduce issue
Start localization_launch.py 
Give Initial pose close to actual location.

#### Expected behavior
Robot will be localized in map.

#### Actual behavior
Robot is localized close to its position then will continue to throw the following error Couldn't determine robot's pose associated with laser scan.

SteveMacenski, 2023-02-01:
Provide the literal error and your configuration please. Can you reproduce with nav2_bringup or on another platform?

Jayden-F, 2023-02-01:
The Error is: Couldn't determine robot's pose associated with laser scan. I will supply a log file ASAP. 

I will also note that the robot I am using is a namespaced turtlebot4.

# This is my current configuration. 

## localization.launch.py 
```
import os

from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node
from nav2_common.launch import RewrittenYaml

def generate_launch_description():

    # rviz_config = os.path.join(get_package_share_directory(
    #     'localization_server'), 'config', 'localization_rviz_config.rviz')
    amcl_yaml = os.path.join(get_package_share_directory(
        'localization_server'), 'config', 'amcl_config.yaml')
    map_file = os.path.join(get_package_share_directory(
        'localization_server'), 'maps', 'box.yaml')

    ARGUMENTS = [
        DeclareLaunchArgument(
            'namespace',
            default_value='',
            description='Robot namespace'),

        DeclareLaunchArgument('use_sim_time', default_value='false',
                              choices=['true', 'false'],
                              description='Use sim time')
    ]

    namespace = LaunchConfiguration('namespace')
    use_sim_time = LaunchConfiguration('use_sim_time')

    param_substitutions = {
        'use_sim_time': use_sim_time}

    configured_params = RewrittenYaml(
        source_file=amcl_yaml,
        root_key=namespace,
        param_rewrites=param_substitutions,
        convert_types=True)

    remappings = [
        ("/tf", "tf"),
        ("/tf_static", "tf_static")]
    # ("/map", "map"),
    # ("/scan", "scan"),
    # ("/amcl_pose", "amcl_pose")]

    map_server = Node(
        package='nav2_map_server',
        namespace=namespace,
        executable='map_server',
        name='map_server',
        output='screen',
        remappings=remappings,
        parameters=[{'use_sim_time': use_sim_time},
                    {'yaml_filename': map_file}]
    )

    amcl_server = Node(
        package='nav2_amcl',
        namespace=namespace,
        executable='amcl',
        name='amcl_server',
        remappings=remappings,
        output='screen',
        parameters=[configured_params]
    )

    lifecycle_manager = Node(
        package='nav2_lifecycle_manager',
        namespace=namespace,
        executable='lifecycle_manager',
        name='lifecycle_manager_localization',
        output='screen',
        parameters=[{'use_sim_time': use_sim_time},
                    {'autostart': True},
                    {'bond_timeout': 0.0},
                    {'node_names': ['map_server', "amcl_server"]}]
    )

    ld = LaunchDescription(ARGUMENTS)
    ld.add_action(map_server)
    ld.add_action(amcl_server)
    ld.add_action(lifecycle_manager)
    return ld

```

## amcl_config.yaml 
```
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_link"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: -1.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    map_frame_id: "map"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan
```

AlexeyMerzlyakov, 2023-02-01:
This message appears when TF can't get "base_link -> odom" transform in your case. One of the root causes of the issue is incorrect bringup setup of your robot, or different times (sim-time/non-sim-time) used by different nodes.
@Jayden-F, could you please provide:
- Are you using real robot or Gazebo simulation of TB4?
- More detailed setup: how did you run the robot? From `localization.launch.py` it seems that you are using your custom `localization_server` package. So, could you please provide full configuration of the system where you are launching, and how are you launching it?
- Get full run log

SteveMacenski, 2023-02-01:
Also note that galactic is EOL at this point. We’d not recommend using it.

Jayden-F, 2023-02-01:
Thank you, @AlexeyMerzlyakov, for your help. I am still learning about ROS and ROS2 and it is likely that the issue is on my end.

I am using a real TB4 robot and have built the packages from these Github repositories:

> Are you using real robot or Gazebo simulation of TB4?

I am using a real TB4 robot and I am building the packages from the following Github repositories:
- https://github.com/Jayden-F/turtlebot4_robot/tree/galactic-namespace
- https://github.com/Jayden-F/turtlebot4/tree/galactic-namespace

> More detailed setup: how did you run the robot? From localization.launch.py it seems that you are using your custom localization_server package. So, could you please provide full configuration of the system where you are launching, and how are you launching it?

System Specs:
Zotac zbox EN1060k
- CPU: Intel® Core™ i5-7500T CPU 
- MEMORY: 8 GB
- GPU: Nvidia GTX 1060-M
- OS: Ubuntu 20.04.5 LTS

I generated a map of my environment using nav2's async_slam_toolbox_node and saved it. To localize my real TB4 robot, I used my custom localization_server package that includes localization.launch.py, amcl_config.yaml, and a map file named box.yaml. The localization.launch.py script starts the map server and AMCL nodes.

However, I experienced an issue where the AMCL localization would only initialize once and then not update. To better illustrate the problem, I created a smaller box map, as my original map was quite large. I have attached the full run log and a screenshot of the issue for further analysis.

>Get full run log

[amcl_64782_1675292920316.log](https://github.com/ros-planning/navigation2/files/10562865/amcl_64782_1675292920316.log)

[image]

SteveMacenski, 2023-02-02:
> Consecutive laser scan transforms failed: (Lookup would require extrapolation into the future.  Requested time 1675292943.069337 but the latest data is at time 1675292943.062120, when looking up transform from frame [base_footprint] to frame [odom]

That's your problem, your odometry (whatever is providing that is not from Nav2) is not publishing data at times when required to process. 

I can't speak to anything TB4 related, I don't maintain or have one.

closed 2023-02-02 by SteveMacenski

Jayden-F, 2023-02-02:
Thank you for your help @SteveMacenski.

I appreciate your support in understanding what's happening. After inspecting the TF tree using tf2_tools view frame, I can see that all the frames are updating as normal. I am also able to drive the robot around and the odom frame updates and moves as expected in rviz2. Is there something I am missing that you could help me understand? I've attached a frames.pdf file and an image of the odom in rviz2.

Thank you for your time!!!

[frames.pdf](https://github.com/ros-planning/navigation2/files/10563606/frames.pdf)

[image]