ros-navigation/navigation2 #3391: We should catch process deaths in the nav2_system_tests, and in those cases fail the test

mmattb, 2023-02-01:
## Bug report

**Required Info:**

- Operating System:
  - All
- ROS2 Version:
  - Rolling
- Version or commit hash:
  - HEAD
- DDS implementation:
  - All
  - 
#### Steps to reproduce issue

Run nav2_system_tests.

#### Expected behavior

If the system test subprocesses like behavior server or planner server die, we should detect that in the system test logic and in that case fail the test if it wasn't intentional.  We should do this to catch shutdown issues, particularly since shutdown issues can kill code cov collection. See Issue #3271. We shouldn't rely on mysterious drops in code coverage for it.

Once we start catching the issues, we can file bugs to go fix them.  There is at least one infrequent issue remaining that I've not been able to figure out.

#### Actual behavior

Processes fail but don't fail the test, so code cov just mysteriously drops. Additionally, shutdown bugs can be obnoxious in various other contexts, such as checking exit codes in shell scripts.

#### Additional information

CC @SteveMacenski 

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2023-02-01:
No argument here. Though, those drops aren't from deaths during the functional tests (that **would** fail the test), they're deaths from the shutdown after functional testing. 

I would tweak the scope to say that we should have a specific test for the clean shutdown of the system that can fail separately. I would not want every functional system test to fail if there was a regression somewhere for maintainer resposibilities. I have to be able to review functional change to the codebase and if during that regression period it would completely block me from being able to merge any significant changes out of fear that a functional test is now failing for other reasons than shutdown cleanliness.

mmattb, 2023-02-02:
Great point!

SteveMacenski, 2023-09-21:
Closing- test still fails either way for us to debug. This is a _Nice to Have_ which won't receive maintainer time.

closed 2023-09-21 by SteveMacenski