ros-navigation/navigation2 #3393: GlobalUpdateGoal and GoalUpdated do not react to updated goals

mrmara, 2023-02-02:
erating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
- Version or commit hash:
  - 1.1.5-1jammy
- DDS implementation:
  - Cyclone

## Feature request

#### Feature description
 The two BT nodes``` GlobalUpdatedGoal``` and ```GoalUpdated``` are responsive to new goals published on ```/goal_pose``` but do not see goal update on ```/goal_update```.
If my understanding of the code is correct they read the current goal from BehaviorTree blackboard but only goals published on ```/goal_pose``` are set inside the blackboard from ```InitializeGoalPose()``` method inside ```navigate_to_pose.cpp```.
This limitation makes the proposed behavior tree in humble ```nav_to_pose_with_consistent_replanning_and_if_path_becomes_invalid.xml``` not responsive to goal update
#### Implementation considerations
The improvement would require minimum changes to the above mentioned nodes or to navigate_to_pose.
For example the nodes or bt_navigator_navigate_to_pose_rclcpp_node could subscribe to the topic and update the blackboard.

SteveMacenski, 2023-02-02:
If you check our docs, you'd see that those are both [condition nodes](https://navigation.ros.org/plugins/index.html) which are simply checking a condition if some new goal has come in via a preempted navigation request or update to the goal. They are used to potentially back out of a current action to replace with a new goal of interest in the behavior trees if a new one comes in. 

New goals can be added via preempting the current action request with a new one, which updates the goal on the blackboard. Its subtle, but separate of the `GoalUpdated` plugins, there is a decorator `GoalUpdater` which will set the blackboard output to the new goal requested over a specific topic at runtime. That blackboard output can be set to the goal blackboard ID set in your configuration file in order to replace a previous goal from the action server and thus trigger the `GoalUpdated` behavior and/or start planning/controlling towards the respective goal. 

So with that explanation:

> The two BT nodes GlobalUpdatedGoal and GoalUpdated are responsive to new goals published on /goal_pose but do not see goal update on /goal_update.

This is not a true statement and an understandable misunderstanding of what those two BT nodes do (and the other `GoalUpdater` node you didn't mention).

> If my understanding of the code is correct they read the current goal from BehaviorTree blackboard but only goals published on /goal_pose are set inside the blackboard from InitializeGoalPose() method inside navigate_to_pose.cpp.

That is not correct. The `goal_update` topic is used by the `GoalUpdader` BT node. Otherwise, the initial goal pose is set from the Action's request goal field. We do have a `goal_pose` topic that can be used instead of the action interface, but it is **highly not recommended** and in fact is never referenced as even existing (I don't think) and exists only for ROS-1-style Rviz plugin requests. Under the hood, that topic callback converts it into an action request which is processed but then the goal handle is dropped on the floor since the user didn't actually want a response or action feedback. This is a really elementary use of the tool and actually problematic if you want to update the goal or ever know if you actually arrived / are done (since you don't have any feedback or response information, as the action gives you). 

But if you load our Nav2 Bringup rviz configs and use the 2D pose tool, that uses the action properly so that it can be preempted with new goals or canceled as your leisure. It looks the same as ROS 1 and the default rviz plugins on purpose. 

So... sounds like you need another BT node in your tree to process goal updates, but also it sounds like you only require that because you're not using the action interface, which you should. If you use the action interface, then you can send a new goal request and it will preempt in the action server side and you shouldn't need to play with the BT at all. You can see this in action when you use our default bringup / BT XML and you can request a goal in rviz2 and then request a new one and it will start navigating to the new goal properly.

closed 2023-02-02 by SteveMacenski

mrmara, 2023-02-03:
Thank you for the detailed explanation. That perfectly solved my problem.
I am sorry for having opened this issue, it was just my misunderstanding of the code.