ros-navigation/navigation2 #3395: [Request] 'One-direction-lane' costmaps

jesusramondovale, 2023-02-03:
I was wondering if is it possible to create a costmap mask (like KeepOut Filters or SlowZones) in order to assure the turtlebot3 won't navigate in a forbidden direction through a corridor, and it'll only travel in the allowed direction? 

I've been trying with GradientLayer mask combined with two KeepOut "walls" around it, but I don't understand quite well what is happening, as the turtlebot3 goes in both directions the same way. 

Thank you very much for any comment or information.

SteveMacenski, 2023-02-03:
I'm sure its possible, but not a feature we offer currently. That could be a good contribution if you were interested in working on it, but might be something better to self host. We are working on a Nav2 Route Server to operate on navigation route graphs which will have directionality embedded which may be a more suitable replacement. Then when following a route, there's no reason for the robot to try to go in the opposite direction so then its just on the local trajectory planner to have bounds placed upon it to stay in the lane (keepouts would be fine)

closed 2023-02-03 by SteveMacenski

jesusramondovale, 2023-02-06:
Hi! It's great to talk directly to you sir! I was thinking a similar solution: use the 'navigate through poses' module and make the 'order sender server' to give the exact coordinates of the entry and the exit of the right corridor, instead of giving only the final goal on the other side of the corridor.
 
But, anyway, does the actual implementation of the costmap grid allow "multidirectional" cost per same pixel pair? I mean: could I code a different cost of going to pixel A to pixel B  than going from B to A with the current data structure of the costmap grid? Can't I, right? 

Thank you for your response!