ros-navigation/navigation2 #3397: Groot doesn't revognize WouldAPlannerRecoveryHelp

muttistefano, 2023-02-06:
Hi All, im on ubuntu 22 - ros rolling.
When trying to open navigate_to_pose_w_replanning_and_recovery.xml with Groot, it complains about "This model has not been registered: WouldAPlannerRecoveryHelp".
I first open the "nav2_tree_nodes.xml" inside the nav2_behaviour_tree package, but i think it lacks the definition of the WouldAPlannerRecoveryHelp node.
Is it normal? 
Thanks

SteveMacenski, 2023-02-06:
It looks like we missed that in @jwallace42 's PR adding the 3 WouldARecoveryHelpXYZ and Are Error Codes Present BT nodes. If you'd be so kind as to open a PR adding those to the Groot XML file, then you should be good to go!

muttistefano, 2023-02-09:
Hi, I created the pull request.
Stefano

closed 2023-02-10 by SteveMacenski