ros-navigation/navigation2 #3399: TurtleBot3 waits for the obstacle to go away

jesusramondovale, 2023-02-08:
**Required Info:**

- Operating System:
    · Ubuntu 22-04 LTS (Desktop)
- ROS2 Version:
    · Humble
- DDS implementation:
  - FastRTPS

  · All configured right, turtlebot3 won't move when an unexpected obstacle is in its way. It just waits the obstacle to leave. Then, the robot starts moving again. 

Can somebody help me?

jesusramondovale, 2023-02-08:
*Edit* 
My nav2_params.yaml local costmap settings: 

local_costmap:
  local_costmap:
    ros__parameters:
      update_frequency: 5.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      use_sim_time: True
      rolling_window: true
      width: 3
      height: 3
      resolution: 0.05
      robot_radius: 0.22
      plugins: ["voxel_layer", "inflation_layer"]
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 3.0
        inflation_radius: 1.5
      voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
        enabled: True
        publish_voxel_map: True
        origin_z: 0.0
        z_resolution: 0.05
        z_voxels: 16
        max_obstacle_height: 2.0
        mark_threshold: 0
        observation_sources: scan
        scan:
          topic: /scan
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "LaserScan"
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: True
      gradient_layer:
        plugin: "nav2_gradient_costmap_plugin/GradientLayer"
        enabled: True
      keepout_filter:
        plugin: "nav2_costmap_2d::KeepoutFilter"
        enabled: True
        filter_info_topic: "/costmap_filter_info"
        
      always_send_full_costmap: True

SteveMacenski, 2023-02-08:
Please ask on ros answers with more specificity.

closed 2023-02-08 by SteveMacenski