ros-navigation/navigation2 #3406: BT plugin rclcpp node not spinning

po1, 2023-02-11:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 18.04
- ROS2 Version:
  - galactic
- Version or commit hash:
  - 1.0.12
- DDS implementation:
  - cyclone

#### Steps to reproduce issue
Launch nav
```
ros2 param list
```

#### Expected behavior
All parameters are listed

#### Actual behavior
```
Exception while calling service of node '/bt_navigatornavigate_through_poses_rclcpp_node': None
Traceback (most recent call last):
  File "/opt/ros/galactic/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.13.1', 'console_scripts', 'ros2')()
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2param/command/param.py", line 39, in main
    return extension.main(args=args)
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2param/verb/list.py", line 116, in main
    sorted_names = sorted(response.result.names)
AttributeError: 'NoneType' object has no attribute 'result'
```

#### Additional information

This seems related to the following questions:
- https://answers.ros.org/question/395435/exception-while-calling-service-of-node-my_node-none/
- https://answers.ros.org/question/387059/ros2-param-list-attributeerror-nonetype-object-has-no-attribute-result/

Admittedly, the `ros2 param` CLI tool could handle the failure in a more graceful way. That said, rclcpp nodes that don't spin seem troublesome whichever way you look at it.

po1, 2023-02-11:
Turns out the `ros2 param` side of this was fixed in humble: https://github.com/ros2/ros2cli/pull/646 

That seems to make this issue somewhat less of a problem. At least I know how to fix my system.

SteveMacenski, 2023-02-11:
Great! I’m not sure there are any action items here for Nav2, am I correct?

po1, 2023-02-11:
If the parameter server in those plugins' rclcpp nodes is the only thing that may be of interest to the outside ROS2 runtime, and those tools are fault-tolerant (like `ros2 param list` is as of humble), then I'd be inclined to say that there probably won't be anything beyond a mild annoyance about those rclcpp nodes not spinning.
So closing this should be fine :)

closed 2023-02-12 by SteveMacenski

mamariomiamo, 2025-03-03:
Sorry to reopen this but may I get some insights about why we keep the ```client_node_``` as non-spinning in the bt_action_server_?
What if we want to change the parameters for individual navigator at run time?

SteveMacenski, 2025-03-04:
Its given to the BT engine to do its own work with as it sees fit (i.e. the BT nodes themselves having a clean, uninterrupted executor to work with that isn't blockable by the server application)