ros-navigation/navigation2 #3407: After a long run, the controller_server crashes.

fig-shimooka, 2023-02-12:
I use Navigation2 to conduct long-distance tests.
After 14 hours, or 6 hours, the controller_server crashed.
This is the gdb error log or this.
It looks like a _Sp_counted_base error.
What causes this.

## Bug report

**Required Info:**

- Operating System:
　Ubuntu 20.04
- ROS2 Version:
   Galactic 

```
optimized out Sp_counted_base^[[0m[INFO] [1676193011.849215801] [controller_server]:  path frame id : map^[[0m
^[[0m[INFO] [1676193012.943634900] [controller_server]:  path frame id : map^[[0m
^[[m--Type <RET> for more, q to quit, c to continue without paging--^M^[[K--Type <RET> for more, q to quit, c to continue without paging-

Thread 1 "controller_serv" received signal SIGSEGV, Segmentation fault.
^[[34m0x00005555555a7e77^[[m in ^[[33mstd::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release^[[m (^[[36mthis^[[m=<optimized out>) at ^[[32m/usr/include/c++/9/bits/shared_ptr_base.h^[[m:148
```

AlexeyMerzlyakov, 2023-02-13:
@fig-shimooka, it would be great if you could provide run logs and GDB backtrace output on this (if you have this one); or automatically collect the backtrace on crash like presented in https://navigation.ros.org/tutorials/docs/get_backtrace.html#automatic-backtrace-on-crash.
This information would definitely help us to understand what is going on and find the clue where the bug could be.

SteveMacenski, 2023-02-14:
Yeah, we need a full traceback as to where that happened to know what happened. As far as I can tell from 5 minutes of googling is that a shared pointer seg faulted on an attempt to reset itself. As far as I know, that didn't even happen within Nav2 and it could have been anything from the controller server, to ROS 2, to DDS, or below without a trace.

SteveMacenski, 2023-02-22:
Any update @fig-shimooka ?

SteveMacenski, 2023-02-27:
Its been over 2 weeks without a response from the original poster without sufficient information to make this actionable. Happy to reopen at a time when either @fig-shimooka is responsive or someone can provide additional information if this is an issue in the server or something in a user's configuration. Given this hasn't been reported by anyone else, I'm willing to think that this isn't a general problem in Nav2 -- but will revisit if other reports or information comes to light.

closed 2023-02-27 by SteveMacenski