ros-navigation/navigation2 #3408: Unmet Dependencies: gz-tools2 : Conflicts: gazebo (>= 11.0.0) but 11.10.2+dfsg-1 is to be installed

Ryanf55, 2023-02-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - rolling
- Version or commit hash:
  - ef4de1527997c3bd813afe0c6296ff65e05700e0

#### Steps to reproduce issue

Try the [build for released distributions](https://navigation.ros.org/build_instructions/index.html#for-released-distributions). During rosdep install, it fails to install `gazebo-ros-pkgs`.

```
ubuntu@ubuntu22:~/Documents/nav2_ws$ rosdep install -y -r -q --from-paths src --ignore-src --rosdistro rolling
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
nav2_bringup: Cannot locate rosdep definition for [slam_toolbox]
Continuing to install resolvable dependencies...
executing command [sudo -H apt-get install -y -qq ros-rolling-gazebo-ros-pkgs]
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y -qq ros-rolling-gazebo-ros-pkgs] failed
  apt: Failed to detect successful installation of [ros-rolling-gazebo-ros-pkgs]
```
```
sudo apt install ros-rolling-gazebo-ros-pkgs
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
The following packages have unmet dependencies:
 gz-tools2 : Conflicts: gazebo (>= 11.0.0) but 11.10.2+dfsg-1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```
#### Expected behavior

I am able to use rosdep to install all necessary dependencies.

#### Actual behavior

Rosdep fails to install dependencies, and I cannot compile on `main`.

padhupradheep, 2023-02-13:
I think gz-tools2 is related to the latest version of Gazebo(formerly known as Ignition) if I’m not wrong.. you need to take up the issue to the folks maintaining Gazebo..

Ryanf55, 2023-02-13:
I was able to clear this up by purging `gz-*`.

```
sudo apt purge gz-*
```

However, this means that I cannot use NAV2 while I have Gazebo garden installed on my system.

padhupradheep, 2023-02-13:
Unfortunately yea! I also faced it. This has to do nothing with Nav2 though. It’s a conflict between Gazebo version. Nav2 at the moment does not support Gazebo Garden