ros-navigation/navigation2 #3410: How are nav2 bts different from groot bts?

ct2034, 2023-02-15:
Hi
When I look at a behavior tree of nav2, for example https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/behavior_trees/navigate_to_pose_w_replanning_and_recovery.xml and open it with groot, save it with groot, it looks very different. 
Why is that? When did groot and nav2 deviate from each other? Why does groot need the additional `TreeNodesModel`?
Can the two formats be mixed?

This of cause may be related to groot itself. I am using BehaviorTree/Groot@4b8f5d3

<details>
  <summary>see examples of the two xmls</summary>

  # before groot

  ```xml
   <root main_tree_to_execute="MainTree">
    <BehaviorTree ID="MainTree">
      <PipelineSequence name="NavigateWithReplanning">
        <RateController hz="1.0">
          <RecoveryNode number_of_retries="1" name="ComputePathToPose">
            <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
            <ReactiveFallback name="ComputePathToPoseRecoveryFallback">
              <GoalUpdated/>
              <ClearEntireCostmap name="ClearGlobalCostmap-Context" service_name="global_costmap/clear_entirely_global_costmap"/>
            </ReactiveFallback>
          </RecoveryNode>
        </RateController>
        <RecoveryNode number_of_retries="1" name="FollowPath">
          <FollowPath path="{path}" controller_id="FollowPath"/>
          <ReactiveFallback name="FollowPathRecoveryFallback">
            <GoalUpdated/>
            <ClearEntireCostmap name="ClearLocalCostmap-Context" service_name="local_costmap/clear_entirely_local_costmap"/>
          </ReactiveFallback>
        </RecoveryNode>
      </PipelineSequence>
    </BehaviorTree>
  </root>

  ```

  # after groot

  ```xml
  <?xml version="1.0"?>
  <root main_tree_to_execute="MainTree">
      <!-- ////////// -->
      <BehaviorTree ID="MainTree">
          <Control ID="PipelineSequence" name="NavigateWithReplanning">
              <Decorator ID="RateController" hz="1.0">
                  <Control ID="RecoveryNode" name="ComputePathToPose" number_of_retries="1">
                      <Action ID="ComputePathToPose" goal="{goal}" path="{path}" planner_id="GridBased" start=""/>
                      <ReactiveFallback name="ComputePathToPoseRecoveryFallback">
                          <Condition ID="GoalUpdated"/>
                          <Action ID="ClearEntireCostmap" name="ClearGlobalCostmap-Context" service_name="global_costmap/clear_entirely_global_costmap"/>
                      </ReactiveFallback>
                  </Control>
              </Decorator>
              <Control ID="RecoveryNode" name="FollowPath" number_of_retries="1">
                  <Action ID="FollowPath" controller_id="FollowPath" goal_checker_id="GoalChecker" path="{path}"/>
                  <ReactiveFallback name="FollowPathRecoveryFallback">
                      <Condition ID="GoalUpdated"/>
                      <Action ID="ClearEntireCostmap" name="ClearLocalCostmap-Context" service_name="local_costmap/clear_entirely_local_costmap"/>
                  </ReactiveFallback>
              </Control>
          </Control>
      </BehaviorTree>
      <!-- ////////// -->
      <TreeNodesModel>
          <Action ID="ClearEntireCostmap">
              <input_port name="service_name">Service name</input_port>
          </Action>
          <Action ID="ComputePathToPose">
              <input_port name="goal">Destination to plan to</input_port>
              <output_port name="path">Path created by ComputePathToPose node</output_port>
              <input_port name="planner_id"/>
              <input_port name="start">Start pose of the path if overriding current robot pose</input_port>
          </Action>
          <Action ID="FollowPath">
              <input_port default="FollowPath" name="controller_id"/>
              <input_port default="GoalChecker" name="goal_checker_id">Goal checker</input_port>
              <input_port name="path">Path to follow</input_port>
          </Action>
          <Condition ID="GoalUpdated"/>
          <Control ID="PipelineSequence"/>
          <Decorator ID="RateController">
              <input_port name="hz">Rate</input_port>
          </Decorator>
          <Control ID="RecoveryNode">
              <input_port name="number_of_retries">Number of retries</input_port>
          </Control>
      </TreeNodesModel>
      <!-- ////////// -->
  </root>
  ```

</details>

SteveMacenski, 2023-02-15:
First off - please don't duplicate. If I believe this needs to involve another maintainer, I can transfer the ticket over but duplicating responses wastes alot of valuable maintainer time to answer your questions 2x by 2 different busy people. No worries, but just for the future.

Groot is a visualization library for behavior trees, a GUI. It stores the behavior trees as XML files like the ones you posted which are then used by Nav2 or other applications based on BT.CPP library (which Groot is attached to). You can visualize any Nav2 BT or any BT.CPP tree in Groot, there is no difference or deviation. Groot is just a visualization tool. 

With that said, we have [a configuration](https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/nav2_tree_nodes.xml) of custom BT nodes which we use to tell Groot about our custom BT nodes -- because BT.CPP was specifically designed so that you can create your own custom BT nodes (which you'd have to do to do _anything useful_ with behavior trees since you need your application's logic). But again, that's not a deviation, we're just telling Groot about the existence and signature of of BT nodes which we can use to develop our applications using BT.CPP. 

Did you load that file? Maybe what you're seeing is that since you didn't load that pallet of known BT nodes, it modified the format to replace things like `ClearEntireCostmap` with `<Action ID="ClearEntireCostmap"` because it is an action node with that name -- however Groot doesn't know anything about a node called `ClearEntireCostmap`.

With that said, if your are loading that XML into Groot so it knows about the nodes and this is still happening, we should let @facontidavide know. As far as I can tell the other changes (e.g. `<?xml version="1.0"?>`) are formatting related and non-problematic.

facontidavide, 2023-02-15:
Some extra information and context:

- It you create an XML manually, open in Groot and save it again, you should expect some things to be different, such as order of attributes, comments, indentation, etc.
- None of these changes should change the semantic of the tree (or it is a serious bug).
- Everything @SteveMacenski said about using models is correct. Since the `TreeNodesModel` is essential, that is appended to the XML to make it self-contained.
- If you work exclusively in Groot, at least opening and saving files should not add any random changes in formatting.

As a reminded, Groot 1 is becoming more and more deprecated and Groot 2 will soon support both BT.CPP 3 and 4 versions (only 4 is currently supported)

ct2034, 2023-02-16:
Thanks for the clarifications @SteveMacenski and @facontidavide.

Let me maybe again summarize my question: 
* I do not think that, semantically, anything changes from opening and saving the file. I just felt that the first version looked like it was a lot easier to look at with just a text editor. And I also think that a diff on that file in a git commit would be better understandable than on the second file, even if it introduced the same change.
* One thing I still don't fully understand is why for example `<PipelineSequence name="NavigateWithReplanning">` becomes `<Control ID="PipelineSequence" name="NavigateWithReplanning">`. But I assume that in the first line it is not even entirely clear that this is a control node, because it needs the additional `TreeNodesModel`. Is this correct? Or are the two formats completely interchangeable? I mean, from it having multiple children it should be clear that it is a control node, I guess.

What I learned:
* When loading the first version of the file, I need to only load the [custom nodes](https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/nav2_tree_nodes.xml), while in the second version I do not need that because the custom nodes are included in the bt xml.

A further question I have:
* I found a schema for the bt xml in https://github.com/BehaviorTree/BehaviorTree.CPP/commit/4c040befa3a6a4af60beb3d6163816d671c213f7. I have not been able to find it in the current repository. How is the scheme defined now? I assume it is just implicit in the code at the moment.

SteveMacenski, 2023-02-16:
We have a tutorial for this you may want to look at https://navigation.ros.org/tutorials/docs/using_groot.html. Its not that you should **only** load that file, but you **need** to load that file (in addition to any others you may have for your own custom nodes) so that groot knows about the custom nodes and what options it provides. 

I can't answer you about the schema, that's a BT.CPP question, not Nav2.

closed 2023-02-21 by SteveMacenski

mamariomiamo, 2023-04-03:
Hi, I am a little bit confused about which version of Groot or which particular commit did Steve use to generate the bt xml files [here](https://github.com/ros-planning/navigation2/tree/main/nav2_bt_navigator/behavior_trees). I also had the same observation as @ct2034. This is what I did:
- Load navigate_w_replanning_only_if_path_becomes_invalid.xml in Groot
- Without any modification, save the tree with another file name
Following difference are observed:
In the original xml file:
> \<ComputePathToPose goal="{goal}" path="{path}" planner_id="SmacPlannerHybrid"/>
In the saved xml file:
> \<Action ID="ComputePathToPose" goal="{goal}" path="{path}" planner_id="SmacPlannerHybrid" server_name="" server_timeout="" start=""/>

When using the new xml file in nav2, following error will be thrown:
> [bt_navigator]: Exception when loading BT: could not initialize rcl action client: failed to get goal service name, at ./src/rcl_action/action_client.c:217

Removing the empty fields: "server_name", "server_timeout" and "start" will solve the problem.

It would be great if @SteveMacenski can clarify if bt xml files in the nav2 repo were generated using Groot.

mamariomiamo, 2023-04-05:
Seems like if Groot2 is used to edit and save the tree, it will be something similar to the original xml file. Fields like server_name or server_timeout can be supplied with default values so that it will be auto-filled when the xml file is saved. This requires modification of the nav2_tree_nodes.xml file.