ros-navigation/navigation2 #3413: BT made with Groot throws error. nav2_tree_nodes.xml possibly wrong

Vitus012, 2023-02-17:
## Bug report
Hello,
I wanted to build a BT for Nav2 with Groot.
I imported the nav2_tree_nodes.xml, like in the tutorial, but it appears to include some ports that cause problems.
For example I want to follow the Object Following tutorial but with Groot.
I add the necessary nodes to the tree, save the BT and want to run it in Nav2 but get the error:

`[component_container_isolated-6] [ERROR] [1676643216.366347825] [bt_navigator]: Exception when loading BT: Possible typo? In the XML, you tried to remap port "service_name" in node [ComputePathToPose / ComputePathToPose], but the manifest of this node does not contain a port with this name.`

This applies to many nodes. It requires me to edit the XML-file afterwards and remove some ports for the BT to load correctly.
This also happens when I import nav2_tree_nodes.xml, then load a working BT. It edits all the nodes to correspond to nav2_tree_nodes.xml, resulting in the BT not working if you save it again.

I think it worked fine a few months ago. I re-downloaded Nav2, Groot and BehaviorTree.CPP but it still throws the same error-message.

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble via apt-get
- Version or commit hash:
  - current Humble branch

#### Steps to reproduce issue
import nav2_tree_nodes.xml in Groot
load working BT (Dynamic Object Following tutorial)
save new BT
ros2 launch nav2_bringup tb3_simulation_launch.py (with saved BT)

#### Expected behavior
BT works like before

#### Actual behavior
`[component_container_isolated-6] [ERROR] [1676643216.366347825] [bt_navigator]: Exception when loading BT: Possible typo? In the XML, you tried to remap port "service_name" in node [ComputePathToPose / ComputePathToPose], but the manifest of this node does not contain a port with this name.`

closed 2023-02-21 by SteveMacenski

Vitus012, 2023-02-21:
Thank you for the answer.
I tried the new file and this BT node is no longer a problem.
I did pull last Friday but I apparently still used the old file.

But when I redid my test I still get an error:
`[component_container_isolated-6] [ERROR] [1677013808.174532477] [controller_server]: FollowPath called with goal_checker name GoalChecker in parameter 'current_goal_checker', which does not exist. Available goal checkers are: general_goal_checker .
`
It appears that the default value for the goal_checker_id for the FollowPath node is no longer accurate.
Simply replacing the GoalChecker default by general_goal_checker seems to be sufficient.
I hope this can be of help.

SteveMacenski, 2023-02-21:
That error seems pretty straight forward - fixed in https://github.com/ros-planning/navigation2/commit/63b690a27a4ba1bde24ebd8e6c592c9894eec169

Vitus012, 2023-02-22:
Sadly I got another error when I tried using Groot while setting up a complete BT from the start.
Since no default is set for the server_name, Groot will set an empty string. (I'm not completely sure, why it didn't do this yesterday, when I basically did the same thing.)
I think since the nodes don't test for an empty string, it will throw the following error, when initializing the action client.

`[bt_navigator-13] [INFO] [1677081507.879261994] [bt_navigator]: Activating
[bt_navigator-13] [ERROR] [1677081508.018218320] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: Error in destruction of rcl action client handle: action client implementation is invalid, at ./src/rcl_action/action_client.c:472
[bt_navigator-13] [ERROR] [1677081508.023881487] [bt_navigator]: Exception when loading BT: could not initialize rcl action client: failed to get goal service name, at ./src/rcl_action/action_client.c:217
[bt_navigator-13] [ERROR] [1677081508.024212559] [bt_navigator]: Error loading XML file:`

I rebuild, just to make sure, but it still happens.

SteveMacenski, 2023-02-22:
Start with our BT trees that work and tell us what's different. I can't guess at Groot's changes or what you changed about the tree to know if its valid

Vitus012, 2023-02-22:
I'm sorry. My comment was a bit too vague.
I wanted to add, that the server_name doesn't have a default in the nav2_tree_nodes.xml, so Groot will just set it as server_name="".
`<Action ID="ComputePathToPose" goal="{updated_goal}" path="{path}" planner_id="GridBased"  server_name="" server_timeout="" start=""/>`
This throws an error, when the node tries to use this empty name for the action client, according to the error message.
If this port is removed from said node in the XML-file, it will work just fine.
`<Action ID="ComputePathToPose" goal="{updated_goal}" path="{path}" planner_id="GridBased" server_timeout="" start=""/>`
I wanted to ask, if maybe the default server_names could eventually be added to the nav2_tree_nodes.xml.

SteveMacenski, 2023-02-22:
Got it- that's an issue for Groot then, it shouldn't be populating fields that were not populated by the users!

robotcopper, 2024-03-13:
Hello, I'm sorry to reactivate this issue but I've been encountering this error for 2 full days:

"[bt_navigator-13] [INFO] [1677081507.879261994] [bt_navigator]: Activating [bt_navigator-13] [ERROR] [1677081508.018218320] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: Error in destruction of rcl action client handle: action client implementation is invalid, at ./src/rcl_action/action_client.c:472 [bt_navigator-13] [ERROR] [1677081508.023881487] [bt_navigator]: Exception when loading BT: could not initialize rcl action client: failed to get goal service name, at ./src/rcl_action/action_client.c:217 [bt_navigator-13] [ERROR] [1677081508.024212559] [bt_navigator]: Error loading XML file:"

and i've only just discovered this issue. my problem is that Groot2 actually creates the ```server_name=""``` and ```server_timeout=""`` fields for me when i don't want them. how can i make sure that Groot2 doesn't create a default value?

Mechazo11, 2025-08-07:
Hi @Vitus012 I ran into this issue today and found a way to solve it

[image]

Keeping it here for future reference