ros-navigation/navigation2 #3421: Failed to call async_cancel_goal in the halt function of BtActionNode to cancel the task

Zhuoqian0, 2023-02-21:
- Operating System: Ubuntu 20.04
- ROS2 Version: Galactic

async_cancel_goal  callbak is like this
```
     auto callback = [&cancel_response_received, this](
        typename ActionT::Impl::CancelGoalService::Response::SharedPtr response)
        {
          RCLCPP_INFO(rclcpp::get_logger("BtActionNode"),  "In async_cancel_goal callback. ");
          if (response) {
            RCLCPP_INFO(
              rclcpp::get_logger("BtActionNode"),
              "In async_cancel_goal callback, response is not nullptr");
          }
            cancel_response_received = true;
        };

      auto future_cancel = action_client_->async_cancel_goal(goal_handle_, std::move(callback));
      // server_timeout_: 500ms
      if (callback_group_executor_->spin_until_future_complete(future_cancel, server_timeout_) !=
        rclcpp::FutureReturnCode::SUCCESS)
      {
        RCLCPP_ERROR(
          rclcpp::get_logger("BtActionNode"),
          "Failed to cancel action server for %s", action_name_.c_str());
      }
```

The judgment condition of the server is like this
```
 while (rclcpp::ok()) {
    if (action_server_ == nullptr || !action_server_->is_server_active()) {
      RCLCPP_DEBUG(get_logger(), "Action server unavailable or inactive. Stopping.");
      return;
    }

    if (action_server_->is_cancel_requested()) {
      RCLCPP_INFO(get_logger(), "Goal was canceled. Stopping the robot.");
      action_server_->terminate_all();
      return;
    }
    UpdateGoal();
}
```

When a problem occurs, ```In async_cancel_goal callback, response is not nullptr``` is printed in the log.But the server doesn't think it needs to cancel the task. ```action_server_->is_cancel_requested() returns false```

Thank you so much！
Good luck！

SteveMacenski, 2023-02-22:
I don't see any code like what you show in your code blocks in the stack. Please point to where this is. If this is application-specific code, I may not be able to help you at a great deal and you should try to reproduce it with our provided BT nodes. If you can reproduce it with something I can access, then we can work through a solution. If you cannot reproduce it, its likely to do with your custom application code (and you'll have a working reference).

Zhuoqian0, 2023-03-20:
Hi,
I haven't tracked this thing down again, and the probability of its recurrence is very low. As the test log you gave above, using the default bringup that we do cancel the actions and they are received by some of the action servers. I will close this question. Thanks you very very much for your attention.

Thanks!
Good Luck!

closed 2023-03-20 by Zhuoqian0