ros-navigation/navigation2 #3422: multi_tb3_simulation_launch: nav2 stack fails at startup

antoineSion13, 2023-02-22:
## Bug report
- Operating System:
  - Ubuntu 20.04 focal
- ROS2 Version:
  - Humble source
- Version or commit hash:
  - 4a7dee1c8c272f2d550b7412f74f1a15cbf25161
- DDS implementation:
  - FastRTPS

#### Steps to reproduce issue

```
ros2 launch nav2_bringup multi_tb3_simulation_launch.py
```

#### Expected behavior
Simulation starts with two turtlebots and two rviz windows. When starting up the Nav2 stack from Rviz, the initial pose can be set for both robots and a goal position can be set. The robots should then follow a path to the goal.
#### Actual behavior
Simulation starts with two turtlebots and two rviz windows. Starting up the Nav2 stack works except for bt_navigator that fails. After that, setting the initial pose fails as well as setting a goal for the robot.
#### Additional information
Log :
[nav2_issue.log](https://github.com/ros-planning/navigation2/files/10803696/nav2_issue.log)

antoineSion13, 2023-02-22:
Update: after commenting these lines in nav2_multirobot_params_1.yaml and nav2_multirobot_params_2.yaml, it works. I do not know why these libraries are not included in nav2 : 

- nav2_are_error_codes_active_condition_bt_node
- nav2_would_a_controller_recovery_help_condition_bt_node
- nav2_would_a_planner_recovery_help_condition_bt_node
- nav2_would_a_smoother_recovery_help_condition_bt_node

SteveMacenski, 2023-02-22:
Thanks for the note - also updated. It looks like I missed something in the sync to Humble a couple of weeks ago. Those were not backported to Humble but somehow a merge resulted in their BT libraries being added to the list in the configuration files.

closed 2023-02-22 by SteveMacenski