ros-navigation/navigation2 #3426: nav2_bringup launch ignores use_sim_time unless it is in params file

nakai-omer, 2023-02-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - humble
- Version or commit hash:
  - ros-humble-navigation2  1.1.5-1jammy.20230124.081803
- DDS implementation:
  - Cyclone-DDS

#### Steps to reproduce issue

1. Do not write `use_sim_time` in nav2_params.yaml
2. Add the following to your launch file
```
IncludeLaunchDescription(
            PythonLaunchDescriptionSource(os.path.join(get_package_share_directory(
                'nav2_bringup') + '/launch/navigation_launch.py')),
            launch_arguments={
                'params_file': os.path.join(pkg_share, 'config/nav2_params.yaml'),
                'use_sim_time': True,
            }.items(),
        ), 
```

#### Expected behavior
`use_sim_time` launch argument should be respected regardless of whether the params file defines it

#### Actual behavior
`use_sim_time` launch argument is ignored, unless params file has an explicit `use_sim_time` config for each of the nodes

#### Additional information
[ This line: ](https://github.com/ros-planning/navigation2/blob/a82db6eebed998414473e34e0be2ab7cb1dc3d45/nav2_bringup/launch/navigation_launch.py#L64)

```
    configured_params = RewrittenYaml(
        source_file=params_file,
        root_key=namespace,
        param_rewrites=param_substitutions,
        convert_types=True)
```
Only replaces use_sim_time in case it exists in params_file, but doesn't add it in case it doesn't.

SteveMacenski, 2023-02-23:
This is a known quirk and was resolved in `main` for Iron release but could not be backported to Humble due to the changes in removing the parameter from the param file, which could break users. We discuss it on our migration guide https://navigation.ros.org/migration/Humble.html#removed-use-sim-time-from-yaml

For Humble and previous distributions, you **must** include `use_sim_time` in the parameter file if you'd like to use the launch argument to set it at runtime -- for the exact reason you mention that we only overwrite existing keys not adding new ones. But if you don't have `use_sim_time` in your param file you can set it via normal ROS 2 mechanisms, but not through our Nav2 launch file provided launch configuration argument. 

I'd just recommend keeping it in the file knowing that it gets overwritten by the launch argument when you set it to whatever for simulation vs hardware use. Else, you can fork the root launch file for your robot (which you probably do anyway) and handle `use_sim_time` yourself if that's not a workable solution. Though, you'll see this go away in future distributions, so that's why my recommendation is just to live with it for now since it won't be a problem for you in the future.

nakai-omer, 2023-02-26:
@SteveMacensk Got it! Thank you so much for the detailed answer.

closed 2023-02-26 by nakai-omer