ros-navigation/navigation2 #3429: Enable failing BT nodes from on_tick()

tonynajjar, 2023-02-27:
## Feature request

#### Feature description
In my BT action and service nodes I do some input validation and would like to fail the node directly, without sending any goal, if the validation fails. I tried calling `halt()` and `setStatus(BT::NodeStatus::FAILURE)` but the goal was still sent.  

#### Implementation considerations

SteveMacenski, 2023-02-27:
I believe you may already for action: https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L196-L201

But you're right services don't have the same param that an `on_tick()` can set to not do. A contribution to add that analog to the Action server node would be a great contribution!

tonynajjar, 2023-02-28:
Ah nice I was on Humble, didn't see that this is already implemented on master. I backported to my branch :+1: 
I opened a draft PR for BTServiceNode

SteveMacenski, 2023-02-28:
OK. I think your other ticket + the PR cover the scope of this ticket, so I'm just going to close it and continue the subconversations in those respective places.

closed 2023-02-28 by SteveMacenski