ros-navigation/navigation2 #3433: tb3_simulation_launch.py: navigation doesn't work.

EGAlberts, 2023-02-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 (jammy)
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  - ros-humble-navigation2	1.1.5-1jammy.20230124.081803
- DDS implementation:
  - Fast-RTPS, I think, whatever is default for my other info.

#### Steps to reproduce issue

```
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False

```
Then give a 2D position, try giving a navigation goal, and try using the Startup button in Rviz.

#### Expected behavior
The turtlebot navigates towards the given goal, as outlined on [Getting Started](https://navigation.ros.org/getting_started/index.html)
#### Actual behavior
Turtlebot doesn't move at all, errors in the terminal.
#### Additional information

I've tried it all in different orders. It seems to either be some dependency on AssistedTeleop? Or more fundamental issue in starting up the behavior tree at all at the beginning. Here is the terminal output:  [nav2_tb3_fails.log](https://github.com/ros-planning/navigation2/files/10851094/nav2_tb3_fails.log)

SteveMacenski, 2023-02-28:
> [component_container_isolated-6] [FATAL] [1677591524.956986977] [behavior_server]: Failed to create behavior assisted_teleop of type nav2_behaviors/AssistedTeleop. Exception: According to the loaded plugin descriptions the class nav2_behaviors/AssistedTeleop with base class type nav2_core::Behavior does not exist. Declared types are  nav2_behaviors/BackUp nav2_behaviors/DriveOnHeading nav2_behaviors/Spin nav2_behaviors/Wait

This seems pretty clear to me. Have you recently updated your binaries and/or need to cleanly re-source your workspace paths? Changes to the plugin indexes require resourcing. Check in the binary path if you can find the assisted teleop plugin in the behavior package. 

As a temporary work around, you can simply remove the assisted teleop from your yaml file's configuration so it doesn't attempt to load it. If you don't use it in your BT anyway, you're not missing out on anything.  I see the assisted teleop package in the last several releases, so it should be there.

EGAlberts, 2023-03-01:
It seems it was an issue with ros2 being outdated indeed. After upgrading everything it works :). So there is some dependency after a certain update of humble for future reference (as in if anyone else has this issue). Thanks a lot!

closed 2023-03-01 by EGAlberts