ros-navigation/navigation2 #3434: Crash when setting should_send_goal_= false in BTActionNode

tonynajjar, 2023-02-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22
- ROS2 Version:
  - rolling
- Version or commit hash:
  - main
- DDS implementation:
  - default

#### Steps to reproduce issue

Checkout https://github.com/tonynajjar/navigation2/tree/test-should-send-goal and try sending a goal from rviz, you will see:

```
[component_container_isolated-6] Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
[ERROR] [component_container_isolated-6]: process has died [pid 69790, exit code -6, cmd '/opt/ros/rolling/lib/rclcpp_components/component_container_isolated --ros-args --log-level info --ros-args -r __node:=nav2_container --params-file /tmp/tmpd_6p7spy --params-file /tmp/launch_params_u1ew9rv6 -r /tf:=tf -r /tf_static:=tf_static'].
```

My branch contain very minimal changes that use this [feature](https://github.com/ros-planning/navigation2/pull/3293), see the diff [here](https://github.com/ros-planning/navigation2/compare/main...tonynajjar:navigation2:test-should-send-goal?expand=1)

#### Expected behavior

No crash

#### Actual behavior

crash

#### Additional information

----

SteveMacenski, 2023-02-28:
Thats not the actual crash report. That’s just map server crashing with your other caused crash. Please provide a traceback

tonynajjar, 2023-02-28:
Edited

SteveMacenski, 2023-02-28:
As a general FYI, if you see that map server lib crash, its almost never that. That's just crashing because something before it crashed. Its a signal to me to look up the log files more. 

I still need a traceback - this is just the logs. This doesn't indicate to me what is wrong.

tonynajjar, 2023-02-28:
> I still need a traceback - this is just the logs. This doesn't indicate to me what is wrong.

Do you mean with backward_ros or gdb? If so, I didn't think that you normally require this as many bug reporters don't know how to get you that.

SteveMacenski, 2023-02-28:
Either or, I even wrote a tutorial about it https://navigation.ros.org/tutorials/docs/get_backtrace.html. I do require it for things that I cannot look at and know the reason (and I don't have the cycles to fish for the problem).

But, as discussed above, its probably due to the fact that the goal wasn't sent so later stuff in the node is assuming some shared pointers are valid when they're not. I think discussing the question about what we would expect to have happen when `should_send_goal_` is `false` sheds some light to the degree at which we need to look into it. If we just return FAILURE, then largely this issue is probably moot

tonynajjar, 2023-03-02:
Hey, sorry this will be delayed a little bit; I did create an internal ticket to tackle this though

closed 2023-03-20 by SteveMacenski