ros-navigation/navigation2 #3436: lifecycle_manager cannot start lifecyce_node inside a namespace

kfabian, 2023-03-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Galactic apt binaries
- Version or commit hash:
  - ros-galactic-navigation2        1.0.12-1focal.20221208.082109
- DDS implementation:
  - Cyclone DDS

#### Steps to reproduce issue
Start a `nav2_util::lifecycle_node` with a namespace set
```python
Node(
    package='nav2_controller',
    executable='controller_server',
    namespace='/my_ns'
    output='screen',
    respawn=use_respawn,
    respawn_delay=2.0,
    parameters=[configured_params],
    arguments=['--ros-args', '--log-level', log_level],
    remappings=remappings + [('cmd_vel', 'cmd_vel_nav')]),
```
And use a lifecycle_manager to startup the node
```python
lifecycle_nodes = ['/my_ns/controller_server',
                   'smoother_server',
                   'planner_server',
                   'behavior_server',
                   'bt_navigator',
                   'waypoint_follower',
                   'velocity_smoother']
...
Node(
    package='nav2_lifecycle_manager',
    executable='lifecycle_manager',
    name='lifecycle_manager_navigation',
    output='screen',
    arguments=['--ros-args', '--log-level', log_level],
    parameters=[{'autostart': autostart},
                {'node_names': lifecycle_nodes}]),
```

#### Expected behavior
The lifecycle_manager configures and activates also the nodes with a namespace set.
#### Actual behavior
The lifecycle_manager will timeout when activating a node with a namespace set, because the bond topic name is not absolute and will also be inside the namespace, e.g `/my_ns/bond`. Remapping the bond topic name to be absolute
```python
Node(
    package='nav2_controller',
    executable='controller_server',
    namespace='/my_ns'
    output='screen',
    respawn=use_respawn,
    respawn_delay=2.0,
    parameters=[configured_params],
    arguments=['--ros-args', '--log-level', log_level],
    remappings=remappings + [('cmd_vel', 'cmd_vel_nav'), ('bond', '/bond')]),
```
will also not work, because `nav2_util::lifecycle_node` uses the node name as the bond id and not the fully qualified node name including the namespace.

SteveMacenski, 2023-03-01:
That seems easy enough a fix:
- https://github.com/ros-planning/navigation2/blob/main/nav2_util/src/lifecycle_node.cpp#L61 add the `/` in front
- https://github.com/ros-planning/navigation2/blob/main/nav2_lifecycle_manager/src/lifecycle_manager.cpp#L190 ditto

That should cover all the task servers (they all use `nav2_util::LifecycleNode`s and the manage itself trying to connect with them.

Can you tell me if that works for you and submit a PR if it does?

kfabian, 2023-03-02:
Thanks for the proposed fix. That solves one half of the problem and should have the same effect as remapping `bond` to `/bond` when starting the node. The other issue is, that I have to configure the lifecycle_manager with the fully qualified node name, so that the correct `change_state` service is used (e.g. `/my_ns/controller_server/change_state`). So in https://github.com/ros-planning/navigation2/blob/8d4f6f4e2dcd37afae94d74e7b5b806ea78e9813/nav2_lifecycle_manager/src/lifecycle_manager.cpp#L190 `node_name` would be `/my_ns/controller_server`, whereas in https://github.com/ros-planning/navigation2/blob/8d4f6f4e2dcd37afae94d74e7b5b806ea78e9813/nav2_util/src/lifecycle_node.cpp#L62 `get_name()` returns `controller_server`, so that the bond connection cannot be created. 
A fix could be to replace `this->get_name()` with `rcl_node_get_fully_qualified_name(this->get_node_base_interface()->get_rcl_node_handle())`. (Unfortunately `LifecycleNode` does not provide `get_fully_qualified_name()`, like a normal node.)

SteveMacenski, 2023-03-03:
But it does have a `[get_namespace](https://docs.ros2.org/foxy/api/rclcpp_lifecycle/classrclcpp__lifecycle_1_1LifecycleNode.html#aa30afa74f1bd249c7c591129db04ebb8)` which can be used to do `get_namespace() + get_name()`. Looking at the implementation though, it should be trivial to add to the lifecycle nodes https://github.com/ros2/rclcpp/blob/432bf21f0261bab209b37ccfe6da550f02751a22/rclcpp/src/rclcpp/node.cpp#L294-L298. I might just do that this afternoon just to knock it out since you bring it up. But either way, you have options that don't involve invoking rcl :smile: 

What I'm curious here is why you're seeing this but we don't see this problem when we're doing multi-robot bringup, which have all their topics and such remapped into namespaces: https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/launch/multi_tb3_simulation_launch.py That's the same thing as you're doing, but at scale of all the nodes.

SteveMacenski, 2023-03-03:
https://github.com/ros2/rclcpp/pull/2115

kfabian, 2023-03-06:
Thanks for having a look at this issue. I think the problem does not happen with the current multi-robot bringup, because the lifecycle_manager is also started with a namespace configured and each robot bringup will have its own bond topic with a namespace. So there is no name clash anymore for the bond ids of the nodes, since they do not share a single bond topic.

SteveMacenski, 2023-03-06:
Got it - if we use a global `/bond` I think it should be fine as well if we have the nodes namespaced there as well as described above. Can you open the PR (after you test to know it works for your needs)?

kfabian, 2023-03-08:
I would like to avoid to open the PR myself, because I would need to request an approval from my employer for contributing to the Nav2 project. The overhead would be disproportionate to the few lines changed.

SteveMacenski, 2023-03-08:
I don't have the extra cycles to make the changes and ensure it works for your particular situation without your participation -- since this isn't an issue I'm running into and small subtle mistakes or overlooking of a detail can cause it not to work and I wouldn't know better. 

You'd have to provide me with either (A) a PR to implement the fix that I can review and approve or (B) a branch which has all the changes you're looking to have tested so I can quickly make the changes I think are necessary and have a test case to know if its fully fixed or not. Either way, you'd need to make some code publicly available (and its probably easier just to go with option A in that case)

> I would need to request an approval from my employer for contributing to the Nav2 project

Think of it as an investment in future contributions :wink:. This project only works when people using it contribute back to it!

SteveMacenski, 2023-03-10:
Changes at my work today have actually made it virtually impossible for me to get to this in the immediately foreseeable future unfortunately. I would very much appreciate if you submitted a patch after going through your internal processes.

kfabian, 2023-03-10:
I totally understand that the only reasonable thing would be that I open a PR for this. Unfortunately I really can not commit on doing it.

SteveMacenski, 2023-03-10:
We may be at an impasse then. I cannot do this for the immediately foreseeable future and even if I were able to, I wouldn't truly know if I resolved your issue without a testing case. That is unfortunate :frowning:

SteveMacenski, 2023-09-13:
@kfabian I opened a PR https://github.com/ros-planning/navigation2/pull/3814 please test / review

SteveMacenski, 2023-09-13:
I'm abandoning the idea of setting a global bond topic, that's incorrect for dealing with multirobot situations in any practical setting. We don't want to have to expose or share that traffic, the bond topic should continue to be namespaced per robot. Anyway, this is not really any part of the issue. 

However the adding of the total node name is doable but causes some problems:

If the namespace is added at launch time(e.g. launch_ros.actions.Node(namespace=my_namespace)` then its not included in the parameterized value sent to the lifecycle manager to process. Then using `get_fully_qualified_name` in the node = `my_namespace/server` while the lifecycle manager only knows about `server`.

So we need to satisfy the following:

- No namespace work
- Namespace set in the launch file works
- Manual namespace for a single thing works

The branch I just pushed should work when none are set or they're set manually but **not** automatically at launch time. The node knows who it is, but the lifecycle manager does not know who the node is properly. The major issue here is that we can't know what namespace is automatically applied and what's manual (or can we?) 

So... thoughts?

kfabian, 2023-09-21:
Thank you very much for looking into this and sorry for the late reply (I have been in vacation). I see your point regarding the multirobot use case and the absolute `/bond` topic name, I just like to point out that I think the `/tf` and `/tf_static` topics have the same issue and they are then remapped to the relative topics in the launch files. Regarding the use of fully qualified node names for the `node_names` parameter I do not completely get why    
```
lifecycle_nodes = [
  namespace + '/controller_server',
  namespace + '/smoother_server',
  namespace + '/planner_server',
  namespace + '/behavior_server',
  namespace + '/bt_navigator',
  namespace + '/waypoint_follower',
  namespace + '/velocity_smoother']
```
is not a feasible solution. 

Nevertheless my desire for using the lifecycle manager to control nodes in a custom namespace was due to a misunderstanding on how the `global_frame` of the controller server is meant to be used, so maybe there is not really a usual use case for this.

SteveMacenski, 2023-09-21:
>  I do not completely get why is not a feasible solution.

Try it locally :smile: It'll throw exceptions and crash in launch. 

The Python Launch API is declarative, so we can't access the actual value of launch configurations to set them until execution time. There may be ways around that, but I haven't found them. Or maybe https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/launch/cloned_multi_tb3_simulation_launch.py#L132 `text substitution` would work? 

But basically the issue is trying to find a clean way to set the namespace in that launch file based on that `namespace` configuration. If we can do that, that would give us at least some path forward. 

Have some cycles to experiment? 

> Nevertheless my desire for using the lifecycle manager to control nodes in a custom namespace was due to a misunderstanding on how the global_frame of the controller server is meant to be used, so maybe there is not really a usual use case for this.

... or perhaps just close it out?

SteveMacenski, 2023-10-05:
Closing out since solution to underlying user problem found. Perhaps there’s a solution to also the technical item as well, but I figure its not worth spending time on a notional problem that impacts no one. Happy to reopen or continue discussion again later if it comes up again!

closed 2023-10-05 by SteveMacenski

kfabian, 2023-10-06:
I agree that it is not worth to spend more time on this. But just for the record, the following syntax for setting the `lifecycle_nodes` parameter inside the launch file would work:
```python
lifecycle_nodes = [(namespace, '/controller_server'),
                   (namespace, '/smoother_server'),
                   (namespace, '/planner_server'),
                   (namespace, '/behavior_server'),
                   (namespace, '/bt_navigator'),
                   (namespace, '/waypoint_follower'),
                   (namespace, '/velocity_smoother')]
```

anaelle-sw, 2023-10-13:
Hello,

We are running into the same issue.

_Context_:
We would like to launch two nav2 `planner_server`, with two different costmaps.
To do this properly, we must use a namespace, to prevent both `planner_server`s's costmaps and actions to share the same name. Using a namespace is currently the only way to rename a `planner_server`'s global costmap.

These are the relevant extracts of our launch file: 
```
    lifecycle_nodes = [
        "controller_server",
        "planner_server",
        "static/planner_server",
        "bt_navigator",
        # ...
    ]

    Node(
      package="nav2_planner",
      executable="planner_server",
      name="planner_server"
      )

    Node(
      package="nav2_planner",
      executable="planner_server",
      namespace="static"
      name="planner_server"
      )
    
    Node(
      package="nav2_lifecycle_manager",
      executable="lifecycle_manager",
      name="lifecycle_manager_nav",
      parameters=[
        {"node_names": lifecycle_nodes},],
      ),
```

--- 

_Tested solutions_ (they all failed for now):
1 - Using the given syntax "`lifecycle_nodes = [('/static',  '/planner_server'), ...]`"

2 - Using `get_fully_qualified_name()` function when creating bond, instead of `get_name()` (and making sure every node name given as parameter to the lifecycle manager starts with "/")

3 - Launch the namespaced planner server with another lifecycle manager, which uses the same namespace

---

We can contribute to solve this issue.

@SteveMacenski  you said that one solution could be:
> The lifecycle manager to try to append the namespace

Could you explain more precisely what you were thinking about?

Do you have any another solution  idea?

SteveMacenski, 2023-10-17:
A couple of questions

- What about just renaming the planner servers to each be their unique names appropriate to their tasks?
- What about remapping the name of the local costmaps? 

>  1 - Using the given syntax "lifecycle_nodes = [('/static',  '/planner_server'), ...]"

Shouldn't the first one of those be `/static/planner_server` to mark the node? Similar to the discussion above, which can/should work if we create a separate function to evaluate the lifecycle nodes that has more of the python context. I wanted to avoid that, but it should be possible (albeit dirty looking)

> 3 - Launch the namespaced planner server with another lifecycle manager, which uses the same namespace

That should work. We already do this with the localization vs navigation lifecycle managers