ros-navigation/navigation2 #3440: [map_saver] Use lossless format for webp

doisyg, 2023-03-02:
## Feature request

#### Feature description
When saving a map in webp format:

> ros2 run nav2_map_server map_saver_cli --fmt webp

It seems that the lossy config is used, it would be nice to save in the lossless format.

(I know webp is not officially supported but it is already working great for loading maps, we just miss saving bit to have full support)

#### Implementation considerations
Really easy with OpenCV, don't really know with GraphicsMagick apart that there is a lossless flag to enable somewhere

SteveMacenski, 2023-03-02:
Png is supported which is lossless, why not use png?

However if you wanted that format, happy to merge a PR to add it.

doisyg, 2023-03-02:
> Png is supported which is lossless, why not use png?

Because webp can be up to 3 times smaller.

> However if you wanted that format, happy to merge a PR to add it.

I may consider it mid term, just wanted to mention it here to not forget. Would it make sense to have a rewrite of the map io stuff with OpenCV ? Main reason I am asking is that I had bad experience in the past with ImageMagick / GraphicMagick dep conflict (and also OpenCV is easier to use for me)

SteveMacenski, 2023-03-03:
I leave that decision to @AlexeyMerzlyakov as the person that's done the most work on the map server stuff in recent history. I don't have a strong preference as long as we're really, really careful not to pull all of openCV into Nav2 as a requirement and only the required IO modules to be installed on a client system.

SteveMacenski, 2023-09-11:
I'm going to close this as a won't fix since our library we use for parsing maps doesn't support this format (yet?). I'm not opposed to having a PR where someone rips it all out and redoes it work a library with more format support natively, but its not something maintainers will focus on given we already support far more options that we used to. 

This is definitely a good contribtion, but not something I feel we should keep in the issue tracker as a topic that we are likely to ever do as a project unless there's a specific client / user that this is sufficiently important to, to implement and submit themselves. 

If however GraphicsMagick does support webp in future versions (seems likely) then you'll get it for free here since the way we use it doesn't rely on the file extensions - we throw it directly into the library to parse. So it very well may just come to you for free in an OS update.

closed 2023-09-11 by SteveMacenski