ros-navigation/navigation2 #3442: BtServiceNode default service name

HovorunB, 2023-03-02:
## Feature request
Hello! I have some ros2 services with custom types and I have a custom BtServiceNode for each one of them. The problem I ran into is that besides the arguments I have to specify the service name every time i use it in the behavior tree, even though the name of the service is always the same.

#### Feature description

Therefore I wanted to ask if you would be interested in adding default service_name which could be passed from BtServiceNode child nodes? I can contribute if needed.

Syntax now:
```
<SetSafetyZone service_name="set_safety_zone" safety_zone="zone1"/>
...
<SetSafetyZone service_name="set_safety_zone" safety_zone="zone2"/>
```
Desired:
```
<SetSafetyZone safety_zone="zone1"/>
...
<SetSafetyZone safety_zone="zone2"/>
```
I also noticed that in the docstring for BtServiceNode describes this parameter, but i think the docstring is wrong and it's merely the BT node name
https://github.com/ros-planning/navigation2/blob/8d4f6f4e2dcd37afae94d74e7b5b806ea78e9813/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp#L42

#### Implementation considerations
Add parameter service_name to BtServiceNode which is an empty string by default, but can be defined from child nodes. This way if nothing is specified in the service_name input port the service node would create client for the default service name

SteveMacenski, 2023-03-03:
> I also noticed that in the docstring for BtServiceNode describes this parameter, but i think the docstring is wrong and it's merely the BT node name

Agreed, that's the node name, I think a time ago it used to be that the node name = the service name and I backed that requirement out because the other contributors didn't understand ROS interface naming conventions.

I'd be open to that contribution :+1: You could make a default value though I don't see where the mechanics of that would work - but that might be a better idea. Or, before we create the service client have a `on_initialization()` function that can be called by the derived node to modify any class members or do initialization work before the client is setup (e.g. change `service_name_`).

SteveMacenski, 2023-03-06:
Approved and imminent merging after a small docs update

closed 2023-03-06 by SteveMacenski