ros-navigation/navigation2 #3447: nav2_util build fail due to rclcpp::ServicesQoS

TZECHIN6, 2023-03-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04.1 LTS x86_64
- ROS2 Version:
  - Humble
- Version or commit hash:
  - ros-humble-navigation2	1.1.6-1jammy.20230221.193247
- DDS implementation:
  - rmw_cyclonedds_cpp

#### Steps to reproduce issue

```
mkdir -p ~/nav2_ws/src
cd nav2_ws/src/
git clone https://github.com/ros-planning/navigation2.git -b humble
cd nav2_ws/
sudo apt update && rosdep update
sudo apt install ros-humble-navigation2 ros-humble-nav2-bringup ros-humble-turtlebot3*
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro $ROS_DISTRO 
colcon build --symlink-install
```

#### Expected behavior
Expected to build successfully as I follow every steps carefully on the Document.
#### Actual behavior
Build fail due to nav2_util package, it throws an error about rclcpp::ServicesQoS
```
--- stderr: nav2_util                              
In file included from /home/thomasluk/nav2_ws/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /home/thomasluk/nav2_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/home/thomasluk/nav2_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp: In instantiation of ‘nav2_util::ServiceClient<ServiceT>::ServiceClient(const string&, const SharedPtr&) [with ServiceT = lifecycle_msgs::srv::ChangeState; std::string = std::__cxx11::basic_string<char>; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>]’:
/home/thomasluk/nav2_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:35:3:   required from here
/home/thomasluk/nav2_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:49:15: error: cannot convert ‘rclcpp::ServicesQoS’ to ‘const rmw_qos_profile_t&’ {aka ‘const rmw_qos_profile_s&’}
   49 |       rclcpp::ServicesQoS(),
      |               ^~~~~~~~~~~~~
      |               |
      |               rclcpp::ServicesQoS
In file included from /opt/ros/humble/include/rclcpp/rclcpp/node.hpp:1321,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/thomasluk/nav2_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:20,
                 from /home/thomasluk/nav2_ws/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /home/thomasluk/nav2_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/opt/ros/humble/include/rclcpp/rclcpp/node_impl.hpp:127:29: note:   initializing argument 2 of ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp::Node::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = lifecycle_msgs::srv::ChangeState; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<lifecycle_msgs::srv::ChangeState> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
  127 |   const rmw_qos_profile_t & qos_profile,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
In file included from /home/thomasluk/nav2_ws/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /home/thomasluk/nav2_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/home/thomasluk/nav2_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp: In instantiation of ‘nav2_util::ServiceClient<ServiceT>::ServiceClient(const string&, const SharedPtr&) [with ServiceT = lifecycle_msgs::srv::GetState; std::string = std::__cxx11::basic_string<char>; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>]’:
/home/thomasluk/nav2_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:36:3:   required from here
/home/thomasluk/nav2_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:49:15: error: cannot convert ‘rclcpp::ServicesQoS’ to ‘const rmw_qos_profile_t&’ {aka ‘const rmw_qos_profile_s&’}
   49 |       rclcpp::ServicesQoS(),
      |               ^~~~~~~~~~~~~
      |               |
      |               rclcpp::ServicesQoS
In file included from /opt/ros/humble/include/rclcpp/rclcpp/node.hpp:1321,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/thomasluk/nav2_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:20,
                 from /home/thomasluk/nav2_ws/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /home/thomasluk/nav2_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/opt/ros/humble/include/rclcpp/rclcpp/node_impl.hpp:127:29: note:   initializing argument 2 of ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp::Node::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = lifecycle_msgs::srv::GetState; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<lifecycle_msgs::srv::GetState> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
  127 |   const rmw_qos_profile_t & qos_profile,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
gmake[2]: *** [src/CMakeFiles/nav2_util_core.dir/build.make:104: src/CMakeFiles/nav2_util_core.dir/lifecycle_service_client.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:210: src/CMakeFiles/nav2_util_core.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< nav2_util [2.40s, exited with code 2]
                                 
Summary: 6 packages finished [4.07s]
  1 package failed: nav2_util
  2 packages had stderr output: nav2_simple_commander nav2_util
  31 packages not processed
```
#### Additional information
I found this issue https://github.com/ros-planning/navigation2/issues/3212 is closed, which has a very similiar error. 

----
## Feature request

#### Feature description

#### Implementation considerations

tonynajjar, 2023-03-06:
Yeah I have the same (and I created the breaking PR). I wonder what's the difference between all our local Humble (or CI environment) and the one from Nav2 CI...

SteveMacenski, 2023-03-06:
Yeah that is odd - I looked and suspect its because we're still using a Rolling base image https://github.com/ros-planning/navigation2/blob/humble/Dockerfile#L9 so we see this particular issue for Humble because of the fact that that's the only API breaking change between Humble and Rolling for the moment in the core stacks. 

Since users don't typically target PRs to Humble (its just me backporting) its not been too much of an issue in the past. I will make a note of it to try to get humble building in CI properly on its own in Humble properly.

closed 2023-03-07 by SteveMacenski

TZECHIN6, 2023-03-08:
Hi @SteveMacenski now the QoS issue is fixed, however I encounter other issue in the humble build. The error is in below:
```
Finished <<< nav2_dwb_controller [1.33s]     
--- stderr: nav2_waypoint_follower           
In file included from /home/thomasluk/nav2_ws/src/navigation2/nav2_waypoint_follower/plugins/photo_at_waypoint.cpp:15:
/home/thomasluk/nav2_ws/src/navigation2/nav2_waypoint_follower/include/nav2_waypoint_follower/plugins/photo_at_waypoint.hpp:37:10: fatal error: cv_bridge/cv_bridge.hpp: No such file or directory
   37 | #include "cv_bridge/cv_bridge.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/photo_at_waypoint.dir/build.make:76: CMakeFiles/photo_at_waypoint.dir/plugins/photo_at_waypoint.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:260: CMakeFiles/photo_at_waypoint.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< nav2_waypoint_follower [1min 24s, exited with code 2]
Aborted  <<< nav2_rotation_shim_controller [29.1s]      
Aborted  <<< nav2_behaviors [3min 10s]                  
Aborted  <<< nav2_smac_planner [4min 1s]                
Aborted  <<< nav2_mppi_controller [4min 28s]
```
The `nav2_waypoint_follower` cannot be built due to lack of cv_bridge package, should I install it manully in the project?

wudiyidashi, 2024-04-02:
All that being said,I refer to the humble branch code to solve it. add .get_rmw_qos_profile() 
`client_ = node_->template create_client<ServiceT>(
      service_name,
      rclcpp::SystemDefaultsQoS().get_rmw_qos_profile(),
      callback_group_);`