ros-navigation/navigation2 #3475: BT navigator node can not load library

hcdiekmann, 2023-03-14:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - ros-foxy-navigation2	0.4.7-1focal.20221104.131617
- DDS implementation:
  - rmw-fastrtps-cpp         1.3.1-1focal.20221012.224708

#### Steps to reproduce issue
Launching `navigation_launch.py`  from own [package](https://github.com/hcdiekmann/pathfinder/blob/main/launch/single_bot_sim.launch.py) with these parameters:

```yaml
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

amcl_map_client:
  ros__parameters:
    use_sim_time: True

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: True

bt_navigator:
  ros__parameters:
    use_sim_time: True
    global_frame: map
[... 235 lines cut ...]
    use_sim_time: True

```

#### Expected behavior
Nav2 bringup launching all nodes.
 
#### Actual behavior
```term
[bt_navigator-9] [INFO] [1678812664.643389759] [bt_navigator]: Configuring
[bt_navigator-9] [ERROR] [1678812664.687773001] []: Caught exception in callback for transition 10
[bt_navigator-9] [ERROR] [1678812664.687802322] []: Original error: Could not load library: libnav2_change_goal_node_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-9] [WARN] [1678812664.687839370] []: Error occurred while doing error handling.
[bt_navigator-9] [FATAL] [1678812664.687855079] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-11] [ERROR] [1678812664.688389597] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-11] [ERROR] [1678812664.688424816] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

hcdiekmann, 2023-03-15:
It is not a custom BT node and the params file was copied directly from the nav2 bringup package [params directory](https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_bringup/bringup/params/nav2_params.yaml). Not sure why I would get this error.

hcdiekmann, 2023-03-15:
Passing the params file path explicitly to `navigation_launch.py` fixed the issue. Although I had adjusted the `params_file` argument in `navigation_launch.py` file to default to that same path. 

As I am launching it together with `slam_toolbox` like this:
```python
    # Load SLAM parameters from yaml file and add node to launch description
    slam_params_file = os.path.join(pkg_pathfinder,'config', 'slam_params_online_async.yaml')
    slam = IncludeLaunchDescription(
                PythonLaunchDescriptionSource([os.path.join(pkg_pathfinder,'launch','slam_online_async_launch.py')]), 
                launch_arguments={'params_file': slam_params_file}.items()
    )

    # Include nav2_bringup launch file
    nav2_params_file = os.path.join(pkg_pathfinder, 'config', 'nav2_params.yaml')
    nav2 = IncludeLaunchDescription(
                PythonLaunchDescriptionSource([os.path.join(pkg_pathfinder,'launch','navigation_launch.py')]),
                launch_arguments={'params_file': nav2_params_file}.items()
    )
    nav2_launch = RegisterEventHandler(
                    event_handler=OnExecutionComplete(
                        target_action=spawn_entity,
                        on_completion=[nav2]
                    )
    )
```
Is it possible that that `navigation_launch.py` used the slam params? As they have the same launch argument names. However the parameters are completely different.

hcdiekmann, 2023-03-15:
Possibly related to [this](https://github.com/ros-planning/navigation2/pull/2243)

SteveMacenski, 2023-03-15:
I’m not sure how it could be related to SLAM. That node name is rather confusing to me since I can’t find any instance of it in the stack so I’m stumped if you didn’t specify it or change anything. 

Your launch comment about not using your file points to using the default file or defaults in the software, both of which I can’t find that BT node lib in the list.

closed 2023-03-15 by SteveMacenski

StetroF, 2023-03-28:
As what [SteveMacenski] say, the [libnav2_change_goal_node_bt_node] in not the action/condition/decorator node provided by navigation2 , So if your navigate_w_replanning_and_recovery.xml has this node but you don't have write it before,the bt navigator would died. If you want to launch your want to launch successfull ,either you delete the relevant node connect with libnav2_change_goal_node_bt_node,or write a libnav2_change_goal_node_bt_node and add this node in your plugin lib