ros-navigation/navigation2 #3476: "demo_security_launch.py" launch file is misnamed in the README tutorial

ionymikler, 2023-03-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22
- ROS2 Version:
  - humble binaries -->
- Version or commit hash:
 - ros-humble-navigation2	1.1.6-1jammy.20230302.224719

#### Expected behavior
Launch of _demo_security_launch.py_
#### Actual behavior
cannot find the launch file
#### Additional information
The actual filename is _security_demo.py_, found it as such in /opt/ros/humble/share/nav2_simple_commander

SteveMacenski, 2023-03-15:
Thanks for the note! I’ll fix that up later today!

SteveMacenski, 2023-03-16:
https://github.com/ros-planning/navigation2/commit/dae2ce65bd5144a05777564b4d46e8109847fdb5

Thanks for pointing this out, fixed!

closed 2023-03-16 by SteveMacenski