ros-navigation/navigation2 #3496: [Collision monitor] Publish `collision_monitor` state on a ROS topic.

kaichie, 2023-03-21:
## Feature request

#### Feature description
Publish `collision_monitor` state on a ROS topic.

Currently, the `collision_monitor` package only prints the Action state on the terminal, it will be useful to publish this as a message so that other parts of the system can use this information.

#### Implementation considerations
The proposed feature should add a new publisher that publishes the state as a ROS topic. Depending on the level of details required, it should at least publish:
- action_type (`<int>` `DO_NOTHING=0`, `STOP=1`, `SLOWDOWN=2`, `APPROACH=3`) 

and/or more of the following fields:

- velocity: the requested velocity that the current state is reporting
- polygon_name: the name of the polygon that is currently activated
- points: the points of the polygon that is currently activated

If more information is required beyond the `action_type`, we probably want to create a new nav2 msgs and also update the `Action` struct to store more of these informations.

SteveMacenski, 2023-03-21:
Sounds reasonable! 

I don't think velocity is as important as the cmd_vel is going to contain that velocity as the normal output. The polygon name might be helpful context, I agree!

tonynajjar, 2023-03-27:
For the msg published by this new node, I would have liked to reuse [CollisionMonitorState](https://github.com/ros-planning/navigation2/blob/70c6ba13ea49e911e9c0fd3c19b33fb9ae4cb965/nav2_msgs/msg/CollisionMonitorState.msg), but in the context of this new node, we can have more than one polygon triggered + we don't care about action_type. Would you recommend creating another msg?

BriceRenaudeau, 2023-04-28:
This issue isn't closed?

closed 2023-05-01 by SteveMacenski