ros-navigation/navigation2 #3508: bt_navigator died unexpectly

StetroF, 2023-03-24:
## Question

**Required Info:**

- Operating System:
  linux 22.04
  - 
- ROS2 Version: 
  humble
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

Here is a part of my launch log file data:
```
1679533271.1928780 [INFO] [behavior_server-27]: process started with pid [81491]
1679533271.1930599 [INFO] [bt_navigator-28]: process started with pid [81493]
1679533271.1932654 [INFO] [waypoint_follower-29]: process started with pid [81495]
1679533271.1934304 [INFO] [velocity_smoother-30]: process started with pid [81497]
1679533271.1969845 [INFO] [lifecycle_manager-31]: process started with pid [81499]
1679533274.4250224 [ERROR] [bt_navigator-28]: process has died [pid 81493, exit code -8, cmd '/opt/ros/humble/lib/nav2_bt_navigator/bt_navigator --ros-args --log-level info --ros-args -r __node:=bt_navigator --params-file /tmp/tmp9whtodqw -r /tf:=tf -r /tf_static:=tf_static'].
```
   As you see ,the bt_navigator died, before this situation happened,  i tried to change the xml file from bt_navigator, what i do is that i copy a file from"navigate_to_pose_w_replanning_and_recovery.xml",and i add some new element in the copy file(change the behavior tree),after that ,i suddenly found that the navigation couldn't work .
    I found this problem when  i want to navigate the car, the navigation in rviz, which the state from Navigation.localization should be active  ,Feedback shoubld be unknown at first,but in my rviz all these three are unknown.
    Base on this problem ,i think it may beause i change the system file in navigation2 so that it can't work.So i first remove all navigation file by
```
sudo apt remove ros-humble-navigation2
```
and i reinstall all of them and nav2_bringup,but even if i reinstall it and start again,the bt_navigator  still died ,What should i do?By the way ,except bt_navigator ,other nodes are all effect

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2023-03-24:
It sounds like your BT is broken. What happens if you just use a BT we provide? Either way, it shouldn't crash, but it would be good to know if its was caused by something due to your BT configuration or due to BT navigator itself to know where to add checks to make sure it doesn't happen. 

If that still doesn't work using a default BT, we need a traceback to see what's up to know how to proceed: https://navigation.ros.org/tutorials/docs/get_backtrace.html

StetroF, 2023-03-28:
I found where is the problem

`bt_navigator:
  ros__parameters:
    use_sim_time: True    ------①It shoube true but i set it false
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    bt_loop_duration: 10
    default_server_timeout: 20
     #'default_nav_through_poses_bt_xml' and 'default_nav_to_pose_bt_xml' are use defaults:
    # nav2_bt_navigator/navigate_to_pose_w_replanning_and_recovery.xml
    # nav2_bt_navigator/navigate_through_poses_w_replanning_and_recovery.xml
    #default_nav_through_poses_bt_xml: "/opt/ros/humble/share/nav2_bt_navigator/behavior_trees/navigate_through_poses_w_replanning_and_re2very.xml"
    default_nav_to_pose_bt_xml: "/home/x/first_ws/src/fishbot-humble/src/fishbot_navigation2/param/copy.xml" ----② i change the 
    default bt_xml to a xml file what a copy 
    plugin_lib_names:`
    after i do these 2 job, the navigation work

closed 2023-03-28 by StetroF