ros-navigation/navigation2 #3511: Feature request: costmap binary filter able to change ros parameters

enricosutera, 2023-03-24:
Hi, what about having the costmap binary filter (or a new one) capable of setting ros parameters along with publishing on a topic?
Some possible use cases: 
- enabling or disabling an obstacle layer source marking or clearing in a specific environment (e.g. infrared proximity sensors in a black environment)
- enabling or disabling full costmap layers (e.g. when going from indoor to pedestrian area).

I realize this can easily be done creating a third part node that subscribes to the currently available topic and modifies the parameter itself or directly having the callback in the node to which the parameter belongs. However I think this would be more flexible and decoupled to have all this is a configuration file!

SteveMacenski, 2023-03-24:
@AlexeyMerzlyakov

enricosutera, 2023-03-27:
I'd  also go with O2 because there would be the flexibility to change parameters also outside the costmap2D.

For what concerns non boolean values, I agree it could be useful, for instance speed limit filter could be replaced as long as max speed is a parameter changeable dynamically. Anyway this could be done later on!

I can work on this btw

AlexeyMerzlyakov, 2023-03-28:
@enricosutera, thank you for taking the interest in this topic! Agree with O2 strategy to use service name as ROS-param for the Binary Filter. The PR is welcoming :blush:

SteveMacenski, 2023-04-27:
Any update? Its been about a month!

enricosutera, 2023-04-28:
Hi! Unfortunately I haven't a solution ready. I had very little time and I have an issue with using the `.../set_parameters` service.
Service calls fail (I did it with rclcpp first, but it's the same with ros2cli):
```
ros2 service call /global_costmap/global_costmap/set_parameters_atomically  rcl_interfaces/srv/SetParametersAtomically   '{parameters: [{name: "static_layer.enabled"}]}'
requester: making request: rcl_interfaces.srv.SetParametersAtomically_Request(parameters=[rcl_interfaces.msg.Parameter(name='static_layer.enabled', value=rcl_interfaces.msg.ParameterValue(type=0, bool_value=False, integer_value=0, double_value=0.0, string_value='', byte_array_value=[], bool_array_value=[], integer_array_value=[], double_array_value=[], string_array_value=[]))])

response:
rcl_interfaces.srv.SetParametersAtomically_Response(result=rcl_interfaces.msg.SetParametersResult(successful=False, reason='cannot undeclare a statically typed parameter'))
```
I have no idea why setting a parameter would require to undeclare it, anyway I think this is related to this old issue https://github.com/ros2/rclcpp/issues/1762 

Also I've tried using `rclcpp::SyncParametersClient` as in the [demos](https://github.com/ros2/demos/blob/rolling/demo_nodes_cpp/src/parameters/set_and_get_parameters.cpp) 
But again I encountered [known issues](https://github.com/ros2/rclcpp/issues/1553).

Naturally It works fine using `ros2 param` so I'll probably take a look at how it operates

AlexeyMerzlyakov, 2023-05-02:
@enricosutera, the reason of such issue is seems that you are note specifying the value of parameter, and this parameter is treated to be unset.
As I understand, the intention is to support boolean `True/False` trigger support which could be specified directly via `boolean` type of ROS-parameter. Something like:
```
ros2 service call /global_costmap/global_costmap/set_parameters_atomically  rcl_interfaces/srv/SetParametersAtomically   '{parameters: [{name: "static_layer.enabled", value: {type: 1, bool_value: False}}]}'
ros2 service call /global_costmap/global_costmap/set_parameters_atomically  rcl_interfaces/srv/SetParametersAtomically   '{parameters: [{name: "static_layer.enabled", value: {type: 1, bool_value: True}}]}'
```
works for me

enricosutera, 2023-05-09:
Thank you @AlexeyMerzlyakov , the issue was really that simple.

As for configuration, does something like this make sense for you?
I'm not sure whether to handle parameters separately or grouped by threshold.

Option 1: custom default state and flip threshold for each parameter
```
      binary_filter:

        plugin: "nav2_costmap_2d::BinaryFilter"
        enabled: True
        filter_info_topic: "/costmap_filter_info"
        transform_tolerance: 0.1
        default_state: False
        binary_state_topic: "/binary_state"
        flip_threshold: 50.0
        parameters: Param0 Param1 Param2
        Param0:
          node_name: "node_0"
          param_name: "param.sub_param"
          default_state: False
          flip_threshold: 50.0
        Param1:
          node_name: "node_1"
          param_name: "param.sub_param"
          default_state: True
          flip_threshold: 25.0
        Param2:
          node_name: "node_2"
          param_name: "param.sub_param"
          default_state: False
          flip_threshold: 75.0
```

Option2: common flip_threshold (in this case one should define two different istances of the filter in case different threshold were desired)
```
      binary_filter:

        plugin: "nav2_costmap_2d::BinaryFilter"
        enabled: True
        filter_info_topic: "/costmap_filter_info"
        transform_tolerance: 0.1
        default_state: False
        binary_state_topic: "/binary_state"
        flip_threshold: 50.0
        parameters: Param0 Param1 Param2
        Param0:
          node_name: "node_0"
          param_name: "param.sub_param"
          default_state: False
        Param1:
          node_name: "node_1"
          param_name: "param.sub_param"
          default_state: True
        Param2:
          node_name: "node_2"
          param_name: "param.sub_param"
          default_state: False
```
An alternative could be grouoing by threshold (that could correspond to map "zones"):

```
      binary_filter:

        plugin: "nav2_costmap_2d::BinaryFilter"
        enabled: True
        filter_info_topic: "/costmap_filter_info"
        transform_tolerance: 0.1
        default_state: False
        binary_state_topic: "/binary_state"
        flip_threshold: 50.0
        parameters: Thresh0 Thresh1
        Thresh0:
          nodes:       [node1,   node1,    node2]
          parameters: [param1,param2,  param1]    
          default_state: False
          flip_threshold: 50.0    
        Thresh1:
          nodes:       [node1,   node2,    node2]
          parameters: [param3,param2,  param3]    
          default_state: False
          flip_threshold: 60.0    

```

AlexeyMerzlyakov, 2023-05-17:
I've checked the local branch. In overall, the approach seems OK for me.
Regarding the testcases, I would make the configuration with `{binary_filter + node1 + node2}`, when `binary_filter` changes `node1` and `node2` binary parameters. The first parameter' defaults value (`node1.binary_param`) might be missed and thus, be equal to filter' default value. Second parameter's default value - to set explicitly (e.g. `setParam(node2.binary_param, False)`). Then check that parameters for these nodes were set correctly. Also, I would like to add the testcases for the situation, when one of the nodes is not running, or not created - to check the `wait_for_service()` verification logic; and the testcase when one of the parameters has no declared (say, send `node1.incorrect_param` change request) - to check errors handling situation.

That is my point of view, so you don't have to strictly follow it, and could propose your scenario of the testcases. The main intention here - is to cover added functionality and logic by new testcases.

enricosutera, 2023-05-25:
I'm ok with what you said:

1. basic flow: change two params from two nodes
2. basic flow 2: test that filter default value is used if no param default is given
3. error control: one node is offline and the other is not -> skip the offline and change the online one
4. error control: target nodes exist but the parameter does not. I expect this to give an error (?)

To be honest you raised a nice "question". One might have made a mistake with params rather than the target node being offline. 
What do you suggest to do in that case? I think best option is to just log it once, at the beginning to easily let find any error.

SteveMacenski, 2023-06-05:
What's the story here? Personally, I think the 3rd party node to take the binary flip to set a parameter makes sense to me. By the time you're setting a bunch of thresholds, that's not really a "binary" filter anymore. It might make sense to simply create a new costmap filter layer to handle param setting on value triggers.

enricosutera, 2023-06-06:
In the end we have only one threshold for the filter, however other than publishing the state in a topic it can change some configured boolean parameters. Maybe I was not clear? 
Generally I don't see great advantages in having another filter just to bridge the boolean topic into services. The only case in which I could find that useful is there's a third part boolean topic, maybe not based on localization (but I don't really have an example in mind)

AlexeyMerzlyakov, 2023-06-07:
> What do you suggest to do in that case? I think best option is to just log it once, at the beginning to easily let find any error.

In both cases: when the target node off-line or when the node has no such parameter declared, the service will return back some kind of failure response. I think, the node need to log an error and continue its work, since the node and/or parameter might be dynamically set later.

> In the end we have only one threshold for the filter, however other than publishing the state in a topic it can change some configured boolean parameters. Maybe I was not clear?

I think, one threshold for the filter should be enough. If you want to have more than one thresholds, you can run multiple instance of binary filters, each own is set it corresponding binary parameters group.

> Generally I don't see great advantages in having another filter just to bridge the boolean topic into services. The only case in which I could find that useful is if there's a third part boolean topic, maybe not based on localization (but I don't really have an example in mind)

I am not sure that understood your intention well. What kind of bridge do you mean? We might have another instance of binary filter, if we want to, say, trigger another binary parameter with another threshold.

enricosutera, 2023-06-08:
> I think, one threshold per a filter should be enough (moreover, it is conceptually correct for the Binary Filter essence). If you want to have more than one thresholds, each one per each parameter, you can run multiple instance of binary filters. Each instance of BF is set it corresponding binary parameter(s) with necessary threshold.

I'm ok to have just one threshold, I was answering to Steve.

> I am not sure that understood your intention well. What kind of bridge do you mean?

I was referring to the 3rd parts node Steve mentioned.

Generally speaking I think the version we've come up with makes sense and it's simple:
- Only one threshold per filter
- Each parameter can have a configurable default value
- Unreachable service or call failures (e.g. wrong param name) lead to log errors but the filter keeps working and will try again later.

AlexeyMerzlyakov, 2023-06-08:
> I was referring to the 3rd parts node Steve mentioned.

I think Steve talked about 3rd party node, as a some other node where we set the variable, in agreement with what was [suggested](https://github.com/ros-planning/navigation2/issues/3511#issuecomment-1550720294).

So, we are should be on the same page, if I understood the intention correctly.

enricosutera, 2023-06-28:
Hello there,

In the end I have these tests:

-  `TEST_F(TestNode, testUnchangedBinaryParams)`
No flip of binary state happens, I check the parameters have not changed. 

- `TEST_F(TestNode, testChangedBinaryParams)`
Flip of binary state happens, I check the parameters have changed.

- `TEST_F(TestNode, testUnsetDefaultBinaryParams)`
Specific default state is not set for target parameters.
Flip of binary state happens, I check the parameters have changed accordingly to filter default state.

- `TEST_F(TestNode, testWrongBinaryParamNode)`
Wrong node name is provided for one of the parameter to be changed.
Flip of binary state happens, I check the parameter of the node with the wrong name has not change, while the other has.
**Shall I had some check on time elapsed here?** Or is the CmakeList.txt timeout enough?

- `TEST_F(TestNode, testWrongBinaryParamName)`
Same as above but instead of the name of the node, the param name itself is wrong.

The test set is [here](https://github.com/enricosutera/navigation2/blob/849a890d92c4767cc0c3cb192b03dd09c29a6ace/nav2_costmap_2d/test/unit/binary_filter_test.cpp#L941)

If that's ok I'm going to create the PR and update the docs.

AlexeyMerzlyakov, 2023-08-10:
@enricosutera, are there any updates on this activity?

enricosutera, 2023-08-12:
Not really, I've done little progress due to lack of time. I hope to give it a last shot in these days

AlexeyMerzlyakov, 2023-08-21:
Yes, please proceed with a new PR to check the concrete code to be reviewed before the merge