ros-navigation/navigation2 #3520: [velocity_smoother] Setting max_accel to negative flips sign of output

Aposhian, 2023-03-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu
- ROS2 Version:
  - humble
- Version or commit hash:
  - https://github.com/fireflyautomatix/navigation2/tree/rpp-dubins-constraint-humble-backport
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
Set 
```yaml
velocity_smoother:
  ros__parameters:
    min_velocity: [0.0, 0.0, -1.0]
    max_accel: [-1.0, 0.0, 0.0]
```

#### Expected behavior
Error to be thrown?

#### Actual behavior
Positive commanded velocity results in negative smoothed velocity, ignoring velocity limits

#### Additional information
I think there are two action items here:
- apply max and min limits last
- throw if signs on accel, decel don't make sense, on both configure, and dynamic parameters callback

SteveMacenski, 2023-03-31:
Sure, seems reasonable. I look forward to the PR :-)

SteveMacenski, 2023-06-07:
@Aposhian any update?

Aposhian, 2023-06-07:
Sorry, fixing this hasn't been a priority for us, and likely won't be for some time.

SteveMacenski, 2023-06-07:
OK, I'll add to the queue to work on this quarter

SteveMacenski, 2023-08-01:
In progress

SteveMacenski, 2023-08-02:
https://github.com/ros-planning/navigation2/pull/3731 Resolves

closed 2023-08-02 by SteveMacenski