ros-navigation/navigation2 #3521: BT_Navigator failing to load Libary: libnav2_change_goal_node_bt_node

USanz, 2023-03-31:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.
- ROS2 Version:
  - Foxy binaries
- Version or commit hash:
  - ca482808a7a7c52ce01ae3c662dc2b980968fc16
- DDS implementation:
  - Eclipse Cyclone DDS

#### Steps to reproduce issue

I've been following [this tutorial](https://www.youtube.com/watch?v=cGUueuIAFgw) in order to have 2 turtlebot robots working with nav2 amcl. So to reproduce that I only downloaded that workspace and launched:
```
ros2 launch turtlebot3_gazebo turtlebot3_tc_world_two_robots.launch.py
```
and
```
ros2 launch localization_server multi_localization.launch.py
```
The problem is that when I try to launch the navigation to make them move to the pose published in rviz (apart from that tutorial)
```
ros2 launch nav2_bringup navigation_launch.py use_sim_time:=true namespace:="tb3_0"
```
I get this error:
```
[ERROR] [1680258456.257513729] []: Caught exception in callback for transition 10
[ERROR] [1680258456.257574221] []: Original error: Could not load library: libnav2_change_goal_node_bt_node.so: cannot open shared object file: No such file or directory
[WARN] [1680258456.257655084] []: Error occurred while doing error handling.
[FATAL] [1680258456.257685068] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
```

#### Expected behavior

The navigation runs correctly

#### Actual behavior

The navigation fails when launching it

#### Additional information

I have also tried deleting the navigation repository to install it from apt but I'm having the same problem.
This seems to be related: #2143

SteveMacenski, 2023-03-31:
Interesting. I thought that was a 1 time user fluke, it looks like there's a real issue there. Can you find a "change_goal_node_bt" anywhere in the code / configuration / etc? I couldn't on Foxy. I have no idea where that is coming from.

USanz, 2023-04-03:
I can't find any "change_goal_node_bt" anywhere in the code nor the config files, but I tried to add it to the plugins filed of the config yaml file with the name "nav2_change_goal_node_bt" as the other plugins and keeps failing. I have also no idea where this error is coming from.

SteveMacenski, 2023-04-05:
Just adding it to the param list is not going to solve the problem. We need to have it **removed** from somewhere.  That doesn't seem to make sense to me that rebooting would cause any such error.  I also cannot find anything with "rotate action" either, nor can I recall ever having those names as BT nodes ever. I don't know where that's coming from. 

If you use **our** launch files, not TB3 or other packages, does it work?

USanz, 2023-04-06:
I've been testing your launch files and I came up with a better result, because it's working for the first robot when publishing a goal, but not for the second:

Executing this command, two rviz2 windows and gazebo opens, with nav2 and amcl  apparently working:
```
TURTLEBOT3_MODEL=waffle ros2 launch nav2_bringup multi_tb3_simulation_launch.py autostart:=True
```

[image]

I see localization active in both rviz windows, but the navigation is only active in one (in the robot1 window) while in the other (robot2, bottom window in the image) is inactive, although autostart is set to True for both. The problem is that when I startup the robot2's navigation it keeps inactive, and if I publish a goal somewhere (in this case in [1.83, -0.85, 0.001]), the robot moves anyway, and it always goes to the coordinates [-0.3, 0.25, 0.001] but it displays the correct coordinates in the console without complaining or giving any error:

```
[rviz2-15] Start navigation
[bt_navigator-23] [INFO] [1680775590.916364345] [robot2.bt_navigator]: Begin navigating from current location to (1.83, -0.85)
[controller_server-20] [INFO] [1680775590.949957867] [robot2.controller_server]: Received a goal, begin computing control effort.
[controller_server-20] [INFO] [1680775592.001315066] [robot2.controller_server]: Passing new path to controller.
```
I've also tried doing it step by step: startup, give a goal and pause the navigation for the robot1 to then try the same with the robot2, just in case they were colliding, but I've got the same result and running it with the autostart:=False the same happens.

SteveMacenski, 2023-04-07:
But the error with the BT nodes goes away? So which thing is causing that?

USanz, 2023-04-18:
Yes, the error with the BT nodes no longer appears, but I've been looking around what could be the problem but I don't see where it is.

Should I open another issue for the navigation problem where one of the robots moves always to the same coords, but not to the given ones?

SteveMacenski, 2023-04-18:
Fix is using the nav2 provided launch files - something appears to be wrong in the tb3 launch files and if it continues to be an issue to address it with them. 

This isn't the right venue for assistance, try on ROS Answers with a great deal of detail.

closed 2023-04-18 by SteveMacenski

danielaffe, 2023-04-18:
??????????? Gesendet mit der mobilen Mail AppAm 18.04.23 um 17:09 schrieb Steve Macenski
                
            
            
                
                    Von: "Steve Macenski" ***@***.***>Datum: 18. April 2023An: "ros-planning/navigation2" ***@***.***>Cc: "Subscribed" ***@***.***>Betreff: Re: [ros-planning/navigation2] BT_Navigator failing to load Libary: libnav2_change_goal_node_bt_node (Issue #3521)
                
                
Fix is using the nav2 provided launch files - something appears to be wrong in the tb3 launch files and if it continues to be an issue to address it with them.
This isn't the right venue for assistance, try on ROS Answers with a great deal of detail.