ros-navigation/navigation2 #3532: the i at the second for loop should be reinit to be zero

dahuacai, 2023-04-17:
**Issue Place:**
    navigation2/nav2_theta_star_planner/src/ThetaStar::initializePosn(int size_inc) 
**raw code:**
`void ThetaStar::initializePosn(int size_inc)
{
  int i = 0;

  if (!node_position_.empty()) {
    for (; i < size_x_ * size_y_; i++) {
      node_position_[i] = nullptr;
    }
  }

  for (; i < size_inc; i++) {
    node_position_.push_back(nullptr);
  }
}`

**Question:**
the i at the second for loop if need reinit to be 0 ?  otherwise the second loop may not work some time and access transgressions can also occur at other place

SteveMacenski, 2023-04-17:
Incorrect - look at the algorithm again. Any element within `node_position_` is set to nullptr if it exists. For the case that there are more requested (e.g. `size_inc > size_x_ * size_y_`) then we need to add in new elements to `node_position_`, which is what's happening in the second loop.

dahuacai, 2023-04-18:
but size_inc = curr_size_y * curr_size_x - size_x * size_y; meanwhile the first for loop size_x,size_y is old . we should reset old size_x * size_y size data to   null and push back new size size_inc, so the  i at second for loop should be re init to be zero ,can we push back the newly added size_inc  num data .if not set to be zero i test happened memory leak. so the second loop should be "for (i = 0; i < size_inc; i++) {"

dahuacai, 2023-04-18:
from line 240 [initializePosn(curr_size_y * curr_size_x - last_size_y * last_size_x);](url) we known that input param size_inc should be incremental data and it may below to size_x_ * size_y_ and it is diff to curr_size_y * curr_size_x

dahuacai, 2023-04-18:
@SteveMacenski

SteveMacenski, 2023-04-18:
Agreed, I did not see that. This looks like it would work for the original initialization (e.g. map size is zero and then becomes non-zero) & at steady state while navigating with localization. I think the issue you bring up would only appear if navigating while performing SLAM so the map is increasing in size over time. 

I read this that it should cause issues when the map is expanding (or contracting, I suppose). Do you see issues in that case which is why you're filing the ticket? That was missing context that there was a detected crash or issue at runtime. 

A PR to resolve with a unit test would be appreciated @dahuacai !

SteveMacenski, 2023-04-19:
Can you open the PR for a fix?

SteveMacenski, 2023-04-27:
Just following up here! I can too, but its always nice to have folks contribute their own work :-)

SteveMacenski, 2023-06-05:
https://github.com/ros-planning/navigation2/pull/3597 resolves

commit 3899db48 (2023-08-02): Resolve #3532: reset i (#3597)
closed 2023-06-05 by SteveMacenski