ros-navigation/navigation2 #3533: [nav2_msgs] Nav2 master build fails after ros rolling update

ehiker, 2023-04-17:
## Bug report

- Operating System:
  - Ubuntu Jammy 22.04
- ROS2 Version:
  - ROS 2 Rolling
- Version or commit hash:
  - nav2 from source, master, SHA ff4fd54076c64f6782b821c60319686413f2551b
  - ros2 rolling apt update & upgrade from 2023-04-17
- DDS implementation:
  - fastrtps and cyclondds

Package nav2_msgs building fails due to strict warning policy in navigation2/nav2_common/cmake/nav2_package.cmake and long-term issue with a redundant declaration in rosidl generation. I reported the issue in more detail in rosidl_typesupport_fastrtps issue, please see https://github.com/ros2/rosidl_typesupport_fastrtps/issues/28

SteveMacenski, 2023-04-17:
What's the error?

SteveMacenski, 2023-04-17:
Huh, neat. I see that as well even after a clean rebuild which usually resolves those kinds of issues after a breaking update.

ehiker, 2023-04-27:
Hi, I can confirm the current master [1cdc277b84115ee97c94b60b2a14630d097c6975](https://github.com/ros-planning/navigation2/commit/1cdc277b84115ee97c94b60b2a14630d097c6975) fixed the issue. I can build nav2_msgs package locally against current ros rolling from APT packages without any workaround.

Thank you.

SteveMacenski, 2023-04-27:
Leaving open since that's a temporary fix, we need to fix rosidl

ehiker, 2023-04-28:
Ok. I am sorry.

SteveMacenski, 2023-04-28:
No need to be sorry :smile:

SteveMacenski, 2023-06-06:
:heart: to @emersonknapp for debugging https://github.com/ros2/rosidl/issues/743#issuecomment-1577716426

closed 2023-06-07 by SteveMacenski