ros-navigation/navigation2 #3535: how can i decrease calculate time on Embedded Chips

dahuacai, 2023-04-19:
**Issuse:**
I use the theta star algorithm on rk3308 embedded chip(with ).when i plan a path with map w,h=(192,254) res=0.05, it wasters near 90ms to get a 15m path.when i use navfn_planner, it only waster 9ms. the 90ms is not acceptable.how can i optimize the code to decrease calculate time.I guess the time was wasted on the loscheck function.Could you give me some suggestions for revision

**plan map:**
[image]

**CPU INFO:**
[image]

dahuacai, 2023-04-19:
@SteveMacenski

dahuacai, 2023-04-20:
that's ok. I will further test and give some feedback

SteveMacenski, 2023-06-02:
@dahuacai any feedback :smile:

SteveMacenski, 2023-06-06:
Closing - happy to reopen or have a PR if you find a way to speed things up! But for now, it meets our requirements on performance. Improvements always welcome, but if there's not a tangible suggestion or actionable change, I don't think we need to have a performance ticket open for every algorithm 'to make it faster'. Obviously, we'd be happy if any algorithm was made faster :smile: 

If you find a potential improvement, I'd love to know!

closed 2023-06-06 by SteveMacenski