ros-navigation/navigation2 #3537: [MPPI] xtensor::optimize not found

kenloumixx, 2023-04-22:
## Bug report

**Required Info:**

- Operating System: ubuntu 18.04
  - 
- ROS2 Version:  humble
  - 
- Version or commit hash: from source
  - 
- DDS implementation: 
  - 

#### Steps to reproduce issue

```
colcon build --symlink-install 
```

#### Expected behavior
Installed
#### Actual behavior
```
[Processing: nav2_mppi_controller]                                   
--- stderr: nav2_mppi_controller                                     
CMake Error at CMakeLists.txt:78 (target_link_libraries):
  Target "mppi_controller" links to:

    xtensor::optimize

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

CMake Error at CMakeLists.txt:78 (target_link_libraries):
  Target "mppi_critics" links to:

    xtensor::optimize

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

```
#### Additional information
I install the xtensor to use the [navigation2 package](https://github.com/ros-planning/navigation2/blob/main/nav2_mppi_controller/CMakeLists.txt#L78:~:text=target_link_libraries(%24%7Blib%7D%20xtensor%20xtensor%3A%3Aoptimize%20xtensor%3A%3Ause_xsimd)) in ROS. 
In the building process, I encounter the above message.
I installed the xtl and xtensor following [the guide](https://github.com/xtensor-stack/xtensor#install-from-sources). However, I don't know why this error occurs. 
Thank you in advance! 

----
## Feature request

#### Feature description

#### Implementation considerations

kenloumixx, 2023-04-23:
Close this issue as this is solved by right installation

closed 2023-04-23 by kenloumixx

baronlq, 2023-05-26:
As I know, [xsimd] is defined as " not available" for my OS version (ubuntu 20.04). Hence, I could not build the "nav2_mppi_controller". I think 22.04 can solve this.

SteveMacenski, 2023-05-26:
Correct, you can install it yourself on 20.04 pretty easily (I did that myself at one point while developing) but there aren't binaries available for some of those packages in older linux OS distributions