ros-navigation/navigation2 #3540: An error was reported while running the foxy-devel branch code

SenHe524, 2023-04-25:
## Bug report
I tried navigation2 by doing the following:
   1.mkdir -p nav2_foxy/src
   2.cd nav2_foxy/src
   3.git clone https://github.com/ros-planning/navigation2.git -b foxy-devel
   4.cd ..
   5.rosdep install --from-paths src --ignore-src -r -y
   6.colcon build --packages-up-to navigation2 --symlink-install --continue-on-error
   7.source install/setup.bash
   8.ros2 launch agv_nav2 agv_nav2.launch.py
      The agv_nav2.launch.py file is as follows:
[image]
   9.The end result is as follows:
[INFO] [launch]: All log files can be found below /home/sen/.ros/log/2023-04-25-15-28-49-234074-ubuntu-196101
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [196106]
[INFO] [map_server-2]: process started with pid [196108]
[INFO] [amcl-3]: process started with pid [196110]
[INFO] [lifecycle_manager-4]: process started with pid [196112]
[INFO] [controller_server-5]: process started with pid [196114]
[INFO] [planner_server-6]: process started with pid [196116]
[INFO] [recoveries_server-7]: process started with pid [196118]
[INFO] [bt_navigator-8]: process started with pid [196143]
[INFO] [waypoint_follower-9]: process started with pid [196162]
[INFO] [lifecycle_manager-10]: process started with pid [196166]
[map_server-2] [INFO] [1682407729.759946865] [map_server]: 
[map_server-2] 	map_server lifecycle node launched. 
[map_server-2] 	Waiting on external lifecycle transitions to activate
[map_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-2] [INFO] [1682407729.760039840] [map_server]: Creating
[amcl-3] [INFO] [1682407729.733463562] [amcl]: 
[amcl-3] 	amcl lifecycle node launched. 
[amcl-3] 	Waiting on external lifecycle transitions to activate
[amcl-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[amcl-3] [INFO] [1682407729.733603926] [amcl]: Creating
[controller_server-5] [INFO] [1682407729.740267506] [controller_server]: 
[controller_server-5] 	controller_server lifecycle node launched. 
[controller_server-5] 	Waiting on external lifecycle transitions to activate
[controller_server-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-5] [INFO] [1682407729.747837880] [controller_server]: Creating controller server
[controller_server-5] [INFO] [1682407729.766808774] [local_costmap.local_costmap]: 
[controller_server-5] 	local_costmap lifecycle node launched. 
[controller_server-5] 	Waiting on external lifecycle transitions to activate
[controller_server-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-5] [INFO] [1682407729.767922347] [local_costmap.local_costmap]: Creating Costmap
[lifecycle_manager-4] [INFO] [1682407729.795385991] [lifecycle_manager_localization]: Creating
[waypoint_follower-9] [INFO] [1682407729.806895284] [waypoint_follower]: 
[waypoint_follower-9] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-9] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-9] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-9] [INFO] [1682407729.807046370] [waypoint_follower]: Creating
[bt_navigator-8] [INFO] [1682407729.809162048] [bt_navigator]: 
[bt_navigator-8] 	bt_navigator lifecycle node launched. 
[bt_navigator-8] 	Waiting on external lifecycle transitions to activate
[bt_navigator-8] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-8] [INFO] [1682407729.809304746] [bt_navigator]: Creating
[lifecycle_manager-10] [INFO] [1682407729.815506392] [lifecycle_manager_navigation]: Creating
[planner_server-6] [INFO] [1682407729.820709179] [planner_server]: 
[planner_server-6] 	planner_server lifecycle node launched. 
[planner_server-6] 	Waiting on external lifecycle transitions to activate
[planner_server-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[lifecycle_manager-4] [INFO] [1682407729.822313892] [lifecycle_manager_localization]: Creating and initializing lifecycle service clients
[planner_server-6] [INFO] [1682407729.824257214] [planner_server]: Creating
[lifecycle_manager-10] [INFO] [1682407729.825371948] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[lifecycle_manager-4] [INFO] [1682407729.830127739] [lifecycle_manager_localization]: Starting managed nodes bringup...
[lifecycle_manager-4] [INFO] [1682407729.830268625] [lifecycle_manager_localization]: Configuring map_server
[lifecycle_manager-10] [INFO] [1682407729.847300428] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-10] [INFO] [1682407729.847361427] [lifecycle_manager_navigation]: Configuring controller_server
[recoveries_server-7] [INFO] [1682407729.853645877] [recoveries_server]: 
[recoveries_server-7] 	recoveries_server lifecycle node launched. 
[... 91 lines cut ...]
[rviz2-1] [INFO] [1682407731.080847533] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [WARN] [1682407731.344320297] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [WARN] [1682407731.367532531] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [WARN] [1682407731.392433696] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [INFO] [1682407731.438793880] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-1] [INFO] [1682407731.439020252] [rviz2]: Sending lifecycle_manager_navigation/is_active request
[rviz2-1] [INFO] [1682407731.439684248] [rviz2]: Waiting for the lifecycle_manager_localization/is_active service...
[rviz2-1] [INFO] [1682407731.439767533] [rviz2]: Sending lifecycle_manager_localization/is_active request
[rviz2-1] [INFO] [1682407731.440999009] [rviz2]: Trying to create a map of size 192 x 329 using 1 swatches
[rviz2-1] [ERROR] [1682407731.445770051] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
[rviz2-1] active samplers with a different type refer to the same texture image unit

   10.Among the notable ones:
[image]

   11.I wonder if this is due to a bug in the source code of the foxy-devel branch that prevents me from running navigation2 correctly

**Required Info:**

- Operating System:
   Ubuntu20.04
- ROS2 Version:
  Foxy binaries
- Version or commit hash:
 foxy-devel

SteveMacenski, 2023-04-25:
>  [bt_navigator-8] [ERROR] [1682407730.131752370] []: Original error: Could not load library: libnav2_compute_path_through_poses_action_bt_node.so: cannot open shared object file: No such file or directory

This is pretty clear, you have the `nav2_compute_path_through_poses_action_bt_node` in your configuration file. Since I can't see your actual yaml file since that's not a public page, I can't point specifically at it. You probably took the yaml file from a more recent branch as your base for use locally when you needed to have taken the one from `foxy-devel`'s branch. You can note in the [foxy branch's file](https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_bringup/bringup/params/nav2_params.yaml#L60) there is no compute path through poses BT node in that parameterization, nor set [as a default](https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_bt_navigator/src/bt_navigator.cpp#L40) in Foxy's params.   

This seems like a clearcut case of user error, most likely :smile:

closed 2023-04-25 by SteveMacenski

SenHe524, 2023-04-26:
Thank you very much for your reply. @SteveMacenski

ibrahimqazi, 2025-01-22:
Hello all, I have the same issue but it reports another node `libnav2_compute_path_through_poses_action_bt_node.so`, it's strange that I installed the Nav2 package from binary not from source which is different than mate @SenHe524 but still this library does not exist. I used the following command to install it.
`sudo apt install ros-foxy-navigation2 ros-foxy-nav2-bringup ros-foxy-turtlebot3*`

@SteveMacenski, do you have any comments on this, please ? Since I am following a book, so at the moment its not possible for me to switch to another distro, it will be time consuming.