ros-navigation/navigation2 #3541: [planner_server][controller_server] Costmaps are Namespaced

AntoBrandi, 2023-04-25:
In the `planner_server` and `controller_server`, when instantiating the Global and the Local costmap they are namespaced

Here the two instructions
https://github.com/ros-planning/navigation2/blob/4aaefad1162d5edcba26878cdb4a2f11322ada99/nav2_controller/src/controller_server.cpp#L63-L71

https://github.com/ros-planning/navigation2/blob/4aaefad1162d5edcba26878cdb4a2f11322ada99/nav2_planner/src/planner_server.cpp#L63-L70

These makes the two nodes have the following names 

```
>>> ros2 node list | grep costmap
/global_costmap/global_costmap
/local_costmap/local_costmap
```

And thus when setting the configuration parameters for the Global and Local costmap we need to set also the namespace

```
global_costmap:
  global_costmap:
    ros__parameters:
      footprint_padding: 0.03
      update_frequency: 1.0
      ...
```

Is there any reason to keep these namespaces? Is it possible to uniform these node's names to the same convention used for the other nodes? 

```
>>> ros2 node list
/amcl
/behavior_server
/bt_navigator
```

And set the parameters without namespacing them?

```
planner_server:
  ros__parameters:
    planner_plugins: ["GridBased"]
    GridBased:
      plugin: "nav2_navfn_planner/NavfnPlanner"
```

SteveMacenski, 2023-04-25:
Unsatisfyingly, I don't have an immediately clear answer. Let me walk you through what I found and lets chat about it and potentially remove it.

I used git blame on the file and found 3 (but really 2 of interest) PRs that play around with this stuff
- https://github.com/ros-planning/navigation2/pull/999 -- Ancient history 
- https://github.com/ros-planning/navigation2/pull/1147 -- Also ancient history, back in the dashing days when we had a world_model node for the global costmap

The first is for adds the idea of having the namespace applied via `absolute_namespace` in the costmap2D API, but is still populated as `nav2_util::add_namespaces(std::string{get_namespace()}, "local_costmap")` in the servers themselves, so it appears similar or the same as today, just not broken apart. The point being for namespacing costmaps for multi-robot applications. 

The second PR's description just says it "fixes" the first's, but I think that's largely just to do with the launch file changes and the API update to pass the parent / local namespaces separately to deal with the `map` topic default. I don't think the impact of the namespacing actually changes here one way or another. 

So that means the first PR is really where this was added meaningfully, and then adjusted for the nuts and bolts later. I'm not finding much particular reason for the second "local_costmap" (or "global_costmap") for the subnamespace myself, but that also seems to not be what is shown in the PR's body as the **topics** being published. The topics aren't under the second namespace at all unless there's a global namespace. Is that true of now as well? 

If so, that could explain point to some subtle issue where nodes within nodes needed something to break them out - though I'm having a hard time remembering anything along those lines myself. I recall a reason, but dashing was a long time ago and ROS 2 was way less stable then, so its very likely any reason for that choice is far outdated. 

I'd be genuinely curious if you looked at the topics and nodes if you remove the last "local_costmap" in 

```
 "local_costmap", std::string{get_namespace()}, "local_costmap", 
``` 

to being a blank line and if there's any difference at all except for the node's name missing the second namespacing. Pay close attention to the costmap publication topics/services and the costmap layer subscription topics. If that all looks good, then I'd run the multirobot launch files too just to make sure those mirror with everything being A-OK. If so, we can remove it.

SteveMacenski, 2023-05-19:
@AntoBrandi any update?

SteveMacenski, 2023-06-02:
@AntoBrandi whats the good word?

AntoBrandi, 2023-06-03:
Hi @SteveMacenski,

thank you for your fast and exhaustive comment. I've tried to remove the last "local_costmap" in

` "local_costmap", std::string{get_namespace()}, "local_costmap",`

with:

` "local_costmap", std::string{get_namespace()}, "",`

but it breaks something. I've not investigated further so I cannot tell exactly which node was complaining about what. I'll try to investigate deeper in the following weeks

SteveMacenski, 2023-08-02:
Ah ok, so from actually now playing with it, I see why we have it this way. 

We have multiple instances of the costmap (e.g. local and global) which we'd like to have its topics namespaced appropriately under the local / global designations (for the footprint, raw costmap, and output costmap + updates). To namespace these topics, we namespace the entire node which throws the node into a redundant namespace of itself. Technically, we could actually make it `global_costmap/costmap` as the node name instead at that point, there's no real reason to have it be `global_costmap/global_costmap` since the namespace makes it unique from the other. 

However, it would still be under _some_ namespace requiring the param file to still have the nested namespaces. 

The question becomes 'how can we set the topics/services of the costmap under the appropriate namespace without namespacing the node itself -- considering that it is an internal node of a larger controller/planner server?' and I don't think that's a super reasonable thing to spend time doing, considering that namespacing a node's **purpose** is to namespace the topics/services/etc within it -- which we want to do. 

The only open question then is what should the names _be_. As you point out,  `global_costmap/global_costmap` is a little odd, but I don't think `global_costmap/costmap` is really any better.

So, in conclusion, I think things should remain as is. The namespace on the node is for the topics / services that would be irritating to individually throw into namespaces manually (and error prune) - as namespaces are intended. The node name could be reverted back to `costmap`, but I don't see that really being a "gamechanger" in the clarity as we discussed above, but comes with the negative that every user needs to update their config files for the new node name which would be alot of effort to make folks go through if there's no tangible improvement that it makes.

closed 2023-08-02 by SteveMacenski