ros-navigation/navigation2 #3545: [bt_navigator] Crashes if lifecycle configure again

BriceRenaudeau, 2023-04-28:
## Bug report
The bt_navigator node crashes when it is restarted using lifecycle process.
```
[INFO] [1682672061.610984122] [lifecycle_manager_navigation]: Configuring bt_navigator
[INFO] [1682672061.611233341] [bt_navigator]: Configuring
[ERROR] [1682672061.612488199] []: Caught exception in callback for transition 10
[ERROR] [1682672061.612506160] []: Original error: parameter 'navigators' has already been declared
[WARN] [1682672061.612526313] []: Error occurred while doing error handling.
[FATAL] [1682672061.612535295] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[ERROR] [1682672061.613942386] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[ERROR] [1682672061.613985222] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
- Version or commit hash:
  - Humble
- DDS implementation:
  - cyclon_dds

#### Steps to reproduce issue
- Start the navigation stack. S
- Stop it using the lifecycle `deactivate` and `cleanup`.
- Restart it using the lifecycle `configure` and `activate`

#### Expected behavior
The node is restarted properly.

#### Actual behavior
The node is not started hence the navigation doesn't work.

#### Additional information
It happens because `declare_parameter()` is called in the `on_configure()` method.
https://github.com/ros-planning/navigation2/blob/1cdc277b84115ee97c94b60b2a14630d097c6975/nav2_bt_navigator/src/bt_navigator.cpp#L140

So when the node is configured again, it crashes on the already declared parameter.

BriceRenaudeau, 2023-04-28:
I will make a PR to fix this.

SteveMacenski, 2023-04-28:
Ah thanks!

closed 2023-04-28 by SteveMacenski