ros-navigation/navigation2 #3547: [nav2_planner] costmap no updated after restart

BriceRenaudeau, 2023-04-28:
## Bug report
The global costmap used by nava2_planner is not updated after a lifecycle restart

- Operating System:
  - Ubuntu
- ROS2 Version:
  - Humble
- Version or commit hash:
  - Humble
- DDS implementation:
  - cyclonedds
  
#### Steps to reproduce issue
```
- Start the navigation stack
- Stop it using the lifecycle `deactivate` and `cleanup`
- Restart it using the lifecycle `configure` and `activate`
```

#### Expected behavior
The nav2_planner is restarted with a working costmap.

#### Actual behavior
The nav2_planner is restarted with a frozen costmap.
The costmap parameters are not accessible, the static_layer is not updated.

#### Additional information
This happens because the costmap thread is deleted in the `on_cleanup()` method and is not recreated in the configure process.
https://github.com/ros-planning/navigation2/blob/1cdc277b84115ee97c94b60b2a14630d097c6975/nav2_planner/src/planner_server.cpp#L255

BriceRenaudeau, 2023-04-28:
There are 2 solutions:
- Don't delete it.
- Create it in the `on_configure()` and not in the constructor.

I am not a fan of the second one as it may change the behavior by moving the creating process in the `on_configure` method.
I will start a PR implementing the first solution.

SteveMacenski, 2023-04-28:
To discuss in the PR since its simple enough, thanks!

closed 2023-04-28 by SteveMacenski