ros-navigation/navigation2 #3549: nav2-smac-planner package not available on foxy

guidout, 2023-04-29:
Sorry if this might not be the right spot for this question.

I'm trying to install the smac-planner package on Foxy but apt install can't find it. Is it not available?

Thank you in advance

SteveMacenski, 2023-04-29:
Its called just the Smac Planner in Foxy https://github.com/ros-planning/navigation2/tree/foxy-devel/smac_planner mostly due to me messing it up. I couldn't change the name back after release so we just have to live with it like that in Foxy unfortunately. I would definitely recommend upgrading to at least Galactic though if you're using the Smac Planner. There are many updates made that were not backported to Foxy.

closed 2023-04-29 by SteveMacenski

guidout, 2023-04-30:
thank you for replying and also for the awesome work you do and share.

SteveMacenski, 2023-04-30:
Yeah

WinstonHChou, 2024-05-06:
Just a quick question, is it possible to build newest smac-planner from source in foxy?
Is there any structure of ros2 that is not supported by ros2 foxy?

tonynajjar, 2024-05-06:
> Is there any structure of ros2 that is not supported by ros2 foxy?

Hard to say but easy to try :slightly_smiling_face:.