ros-navigation/navigation2 #3550: Error messages with 'undefined symbol'

kenloumixx, 2023-04-29:
## Bug report

**Required Info:**

- Operating System: ubuntu 18.04
  - 
- ROS2 Version: humble
  - 
- Version or commit hash: from source
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```
In dusty-nv container, I build the navigation2 and related packages.
The build ends successfully but when I launch the program with navigation2, the following error occurs.
```

#### Expected behavior
Run successfully 
#### Actual behavior
`/root/ros2_humble/install/lib/robot_state_publisher/robot_state_publisher: symbol lookup error: /root/ros2_humble/install/lib/robot_state_publisher/robot_state_publisher: undefined symbol: _ZNK12class_loader11ClassLoader14getLibraryPathB5cxx11Ev`
`[nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libcontroller_server_core.so [component_container_isolated-3] [ERROR] [1682701839.979859721] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libcontroller_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701839.983249149] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libmap_server_core.so [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'controller_server' of type 'nav2_controller::ControllerServer' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libcontroller_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [ERROR] [1682701840.003610261] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libmap_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'map_server' of type 'nav2_map_server::MapServer' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libmap_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.056172602] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libsmoother_server_core.so [component_container_isolated-3] [ERROR] [1682701840.083442431] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libsmoother_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.095752612] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libamcl_core.so [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'smoother_server' of type 'nav2_smoother::SmootherServer' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libsmoother_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [ERROR] [1682701840.116262721] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libamcl_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'amcl' of type 'nav2_amcl::AmclNode' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libamcl_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.144925078] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libplanner_server_core.so [component_container_isolated-3] [ERROR] [1682701840.150345071] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libplanner_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.162698710] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libnav2_lifecycle_manager_core.so [component_container_isolated-3] [ERROR] [1682701840.170959088] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libnav2_lifecycle_manager_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'planner_server' of type 'nav2_planner::PlannerServer' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libplanner_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'lifecycle_manager_localization' of type 'nav2_lifecycle_manager::LifecycleManager' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libnav2_lifecycle_manager_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.201203546] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libbehavior_server_core.so [component_container_isolated-3] [ERROR] [1682701840.216321887] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libbehavior_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'behavior_server' of type 'behavior_server::BehaviorServer' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libbehavior_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.229592645] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libbt_navigator_core.so [component_container_isolated-3] [ERROR] [1682701840.244705002] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libbt_navigator_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'bt_navigator' of type 'nav2_bt_navigator::BtNavigator' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libbt_navigator_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.254544315] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libwaypoint_follower_core.so [component_container_isolated-3] [ERROR] [1682701840.259580679] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libwaypoint_follower_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'waypoint_follower' of type 'nav2_waypoint_follower::WaypointFollower' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/instal [... cut ...] l/lib/rviz2/rviz2: symbol lookup error: /root/ros2_pkgs/install/lib/librviz_common.so: undefined symbol: _ZNK12class_loader23MultiLibraryClassLoader27getAllAvailableClassLoadersEv [ERROR] [rviz2-2]: process has died [pid 20351, exit code 127, cmd '/root/ros2_pkgs/install/lib/rviz2/rviz2 -d /workspace/install/share/a1_robot/rviz/nav2_default_view.rviz --ros-args -r __node:=rviz2']. `
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2023-04-29:
Unfortunately, we do not offer support for 18.04 anymore, please upgrade to a more recent distribution. There hasn't been a ROS 2 distribution using 18.04 in a couple of years. 

At its core, it looks like class loader isn't able to find components to load them into the containers. That's not really a Nav2 problem, but a class loader problem (for which composition containers launching Nav2's nodes are using). It seems like you're using the `main` branch or one of the more recent distribution branches, so you may want to try using a distribution branch of Nav2 that's more in line with the modernity of your OS / ROS 2 version. Those will pre-date Nav2 using composition so that may solve your problem.

closed 2023-04-29 by SteveMacenski

kenloumixx, 2023-05-08:
@SteveMacenski Thank you for the answer. I used the Nav2 humble branch which is in line with my ROS2 version. In this case, does this matter?