ros-navigation/navigation2 #3558: Simple Commander API - BasicNavigator - GetPath | planner_id and use_start get overriden

nakai-omer, 2023-05-07:
**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble     
- Version or commit hash:
  - ros-humble-navigation2	1.1.6-1jammy.20230327.054544
- DDS implementation:
  - cyclone-dds

#### Steps to reproduce issue
https://github.com/ros-planning/navigation2/blob/9b56556dfd9c50f4046d253317cd0949a4f34583/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#L353

You can see in above line that parameters are being overridden by default values, causing planner_id and use_start to get the default values.

#### Expected behavior
`getPath` should use the parameters

#### Actual behavior
`getPath` overrides them with default values

nakai-omer, 2023-05-07:
Added a PR to fix this and a couple of more places with the same bug

SteveMacenski, 2023-05-07:
Will merge shortly!

closed 2023-05-07 by SteveMacenski

nakai-omer, 2023-05-08:
@SteveMacenski Also added a PR for humble branch #3562

PR #3562 (2023-05-09): Fix wrong usage of default params in python3 commander API
---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | #3558 |
| Primary OS tested on | Ubuntu |
---

## Description of contribution in a few bullet points
Fixed #3558 for Humble branch as well

#### For Maintainers: 
- [ ] Check that any new parameters added are updated in navigation.ros.org
- [ ] Check that any significant change is added to the migration guide
- [ ] Check that any new features **OR** changes to existing behaviors are reflected in the tuning guide
- [ ] Chec [... cut]