ros-navigation/navigation2 #3560: Lifecycle Manager and Nodes not communicating

Jayden-F, 2023-05-08:
## Bug report

**Required Info:**

- Operating System:
  
  - Ubuntu 22.04
- ROS2 Version:
  
  - Humble binaries
- Version or commit hash:
  
  - ros-humble-navigation2	1.1.6-1jammy.20230426.153445

- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

``` 
ros2 launch nav2_bringup navigation_launch.py
```

#### Expected behavior
Nodes are started by lifecycle manager

#### Actual behavior
Node and lifecycle manager are waiting for each other.

#### Additional information

Logs:

Lifecycle Log
~~~
[INFO] [1683508464.962831486] [lifecycle_manager_navigation]: Creating
[INFO] [1683508464.964380219] [lifecycle_manager_navigation]: [34m[1mCreating and initializing lifecycle service clients[0m[0m
[INFO] [1683508466.965240083] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[INFO] [1683508468.965448294] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[INFO] [1683508470.965534730] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[INFO] [1683508472.965671265] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[INFO] [1683508474.965809085] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[INFO] [1683508476.966011392] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[INFO] [1683508478.966254587] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[INFO] [1683508480.966423796] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[INFO] [1683508482.966664262] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[INFO] [1683508484.339379613] [rclcpp]: signal_handler(signum=2)
[INFO] [1683508484.341005065] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[INFO] [1683508484.341069223] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
~~~

Controller Log
~~~
[INFO] [1683508464.910861968] [controller_server]: 
	controller_server lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [1683508464.930517454] [controller_server]: Creating controller server
[INFO] [1683508464.943213765] [local_costmap.local_costmap]: 
	local_costmap lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [1683508464.944312065] [local_costmap.local_costmap]: Creating Costmap
[INFO] [1683508484.339472185] [rclcpp]: signal_handler(signum=2)
~~~
All Logs
[Logs.zip](https://github.com/ros-planning/navigation2/files/11416973/Logs.zip)

SteveMacenski, 2023-05-08:
Have you tried switching RWMs to Cyclone DDS? Could be a regression in services discovery in Fast-DDS (which has been known to happen, so its my first intuition to try that)

Jayden-F, 2023-05-09:
Thank you for getting back to me. I have now validated it works with Cyclone DDS. Unfortunately, as I am working with multiple robots the quadratic network traffic associated with Cyclone DDS is prohibitive. 

Do you have further details regarding the Fast-DDS regression, so that I can correctly create an issue against ePromisa/Fast-DDS.

nirwester, 2023-05-10:
I can confirm that services, on Fast-DDS, are not working properly (at least they didn't a couple of months ago), while they do using Cyclone.

SteveMacenski, 2023-05-10:
CC @EduPonz from eProsima for a bug report

SteveMacenski, 2023-05-10:
Thanks! Let me know if there's anything I can do to help facilitate

MiguelCompany, 2023-05-10:
With the default configuration, and running on the official ubuntu docker image, I cannot reproduce this.

I also installed rmw-cyclonedds-cpp so I could compare both executions.

```bash
docker pull ros:humble
docker run --rm -it ros:humble

apt update
apt install ros-humble-nav2-bringup
apt install ros-humble-rmw-cyclonedds-cpp
RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 launch nav2_bringup navigation_launch.py
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 launch nav2_bringup navigation_launch.py
```

[log_cyclonedds.zip](https://github.com/ros-planning/navigation2/files/11440875/log_cyclonedds.zip)
[log_fastdds.zip](https://github.com/ros-planning/navigation2/files/11440876/log_fastdds.zip)

Jayden-F, 2023-05-10:
For fast dds, I am running the discovery server on a Turtlebot4 over the network. The configuration is in the following repo: https://github.com/turtlebot/turtlebot4_setup/tree/humble/turtlebot4_discovery

MiguelCompany, 2023-05-10:
@Jayden-F That is what I supposed you were doing, as I commented [here](https://github.com/eProsima/Fast-DDS/issues/3505#issuecomment-1541911457)

I wanted @SteveMacenski and other users to be aware that I checked with the default configuration and I saw no issues.

I think the title of this issue might be misleading. I would add something like ` on Turtlebot4 with Discovery Server` at the end.

SteveMacenski, 2023-05-10:
Though, if it doesn't work for a TB4, I'm not sure that the setup for any particular applied robotics application would be much different - but I haven't looked in detail as to what the settings they setup for the TB4 are

JaimeMartin, 2023-05-19:
@SteveMacenski, if you read this comment:

https://github.com/eProsima/Fast-DDS/issues/3505#issuecomment-1541933623

You can see they have problems with cycloneDDS, just with a couple of Robots, and they have dozens, which is why they are using Fast DDS and its discovery server.

Jayden-F, 2023-05-19:
@JaimeMartin That is correct. Unfortunately, I cannot start the navigation stack using Fast DDS and its discovery server for even one robot.

JaimeMartin, 2023-05-19:
Hi @Jayden-F ,

I don't know about your configuration (actually is weird you are having problems with both CycloneDDS and Fast DDS + discovery server in such with just 1-2 nodes).

You can see there are robots out there using Navigation and ROS 2 humble with Fast DDS and combinations of our ROS2 Discovery server and the ROS2 Router:

https://vulcanexus.org/news/vulcanexus-cloud-communication-in-distributed-ros-2-networks

So, sure, it works. Could you please send an email to our support team to organize a meeting with you guys?

Jayden-F, 2023-05-19:
@JaimeMartin Thank you for looking into my issue. I will send an email to organise a meeting in the meantime. The issues I am experiencing with CycloneDDS differ from those with Fast DDS. Additionally, there are a considerable amount of nodes in the setup. unfortunately, I do not have a rqt screenshot. However, this image shows some of the nodes associated with just nav2.

[image]

With Cyclone, I can use the navigation stack as the video demonstrates for one robot. However, things go pear-shaped with multiple robots on the same domain_id and namespacing to separate them. Due to cyclones multicasting the network traffic dramatically increases. I have recently had success separating the robots on different domain ids and have been able to control 2 robots with a navigation stack. 

With FastDDS, The ROS Services do not communicate at all, so the navigation stack lifecycle nodes are both waiting for each other to be ready and do not progress. However, works with cyclone.

Let me know if you need further information.