ros-navigation/navigation2 #3561: Costmap width and height should be treated as double

lukicdarkoo, 2023-05-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
- Version or commit hash:
  - main
- DDS implementation:
  - CycloneDDS

## Feature request

Is there a good reason why the width and height costmap parameters are restricted to be an integer?
https://github.com/ros-planning/navigation2/blob/356e385e06afd43759a7718c7704d7a6995f3086/nav2_costmap_2d/src/costmap_2d_ros.cpp#L386

I briefly checked the implementation and there is no obvious reason why this is restricted to int. I would make a PR to allow the width and height parameters to be double if you agree. There are some specific use cases in which it would be quite useful, e.g. for small robots we need a greater granularity or we can use it to cut out the map.

SteveMacenski, 2023-05-08:
I'm torn between this not being a big deal so its easy enough to allow & this would also make it so that every user of Nav2 would have to change their configuration files, without fail, or have launch-time failures. There's no good way to depreciate gradually the use of one or another since we have to explicitly declare its type. 

A meter is not a good granularity but I was looking back at ROS 1 docs and it appears to also be in the `int` type for whole number of meters. This might be something good to post for a discussion point on Discourse and see if there are any strong opinions for/against before deciding what to move forward with.

SteveMacenski, 2023-05-10:
Any interest in pushing this forward with a public opportunity for comment on Discourse?

SteveMacenski, 2023-06-05:
Closed due to inactivity and that this has been the long-term status of the param configuration since 2009. If there's a strong desire to have this changed, we'll discuss it then, but I don't see a reason to fix what aint broke from users to have requested it until now (and the passive nature of this request doesn't indicate its all that important to you either)

closed 2023-06-05 by SteveMacenski